christianhaitian/arkos #617: RG353M Bluetooth shows numbers/letters instead of actual device names

Raiden85OCUK, 2023-01-18:
Other than this bug everything seems to be working great, even the wifi is ok.

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

When you try to connect to a new bluetooth device, after it scans for available devices it shows all the device names as numbers/letters instead of their actual names so I have no idea what device it is. 

### How can the issue be reproduced? (If applicable)

Happens every time I open the bluetooth settings and it starts scanning. Though one device seems to be read correctly for some reason.

### What device are you using?

RG353M 

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

v2.0_01152023

[image]

christianhaitian, 2023-01-18:
This means the device is not showing it's name via bluetooth.  Usually this is fixed by just going back to the main menu of the bluetooth manager and pressing connect to new device again.  Also, it seems you have quite a few bluetooth devices in your area, you're sure your device just wasn't available at the very bottom of the menu?  That menu was only showing about a 1/3 of the bluetooth devices it found in your area.  You may need to scroll down by tapping the down button repeatedly.

Raiden85OCUK, 2023-01-18:
Thanks for that, didn't realise it won't show the actual name in the list until I put that particular device into pairing mode then the actual name shows up fine, my mistake as I hadn't seen a device do that before as they either show the name or nothing at all. Glad it's working as intended👍

closed 2023-01-18 by Raiden85OCUK