christianhaitian/arkos #620: New Anbernic RG353V won't start up with

scspijker, 2023-01-19:
### Describe the Issue (If applicable)
I have bought a brand new Anbernic RG353V. Received it today. Boots up original OS just fine. Boots up Android just fine.
Also bought / received a Samsung Evo Plus 64GB SD card.
Written `ArkOS_RG353V_v2.0_01152023.img` to SD card using ApplePiBaker. Waited until completion.

Put SD card into unit, turned unit on. Unit shows white screen with black vertical bar. Let the unit sit for ~5 minutes, it rebooted twice (like normal, expanding partitions and stuff?). Still shows white screen with black vertical bar. 

Turned unit off with Function + Power button.
Turned unit on.
Same white screen + black bar.

Went back to original OS SD card, booted up fine.
Written the ArkOS.img to the original OS SD card. Again the white screen with black vertical bar.

Tried an older version, `ArkOS_RG353V_v2.0_12302022.img`. Again, white screen with black vertical bar.

Problems with display driver? New hardware revision from Anbernic?

### How can the issue be reproduced? (If applicable)
Write `ArkOS_RG353V_v2.0_01152023.img` to an SD card, try to boot up a recently acquired Anbernic RG353V with it.

### What device are you using?
Anbernic RG353V

### What version of ArkOS are you on?
ArkOS_RG353V_v2.0_12302022
ArkOS_RG353V_v2.0_01152023

### Anything else you'd like to include that may help to review this issue or feature request?
Photograph:
[image]

christianhaitian, 2023-01-19:
Looks like you have one of the units with the updated screen.  None of the custom firmwares have a good way to provide simple support for it just yet.  For now, you’ll have to reimage the sd card, then upon completion, go into the boot partition(accessible from Mac, Windows, or linux), delete the file named rk3566-OC.dtb, copy the file that’s named rk3566-OC.dtb.v2 to rk3566-OC.dtb then boot your device.

christianhaitian, 2023-01-19:
If this works for you, please let me know.  Also, it would be helpful if you can provide a screenshot of the back of your unit.  I want to see if there are any version markings on the unit or any other visual indicators that the unit is different besides that garbled screen on boot.

scspijker, 2023-01-20:
I will try tonight!
I'll screw open the unit to get some version numbers. Thanks!

scspijker, 2023-01-21:
@christianhaitian Here you are, little bit late, sorry :)

Rear
[image]

Rear inside
[image]

Front inside, with battery and BT/Wifi antenna detached and backing plate for analog sticks removed
[image]

Front side of PCB
[image]

Below PCB on front side of casing
[image]

Numbers on display cables
[image]

Nothing below display cables
[image]

Cheers, hope you can make something out of this.

scspijker, 2023-01-21:
@christianhaitian Sadly I was unable to access the boot partition after writing the image to SD card using "ApplePiBaker".

```
❯ diskutil list /dev/disk6
/dev/disk6 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *64.1 GB    disk6
   1: A60B0000-0000-4C7E-8000-015E00004DB7               4.2 MB     disk6s1
   2: D46E0000-0000-457F-8000-220D000030DB               4.2 MB     disk6s2
   3: 4F4C0000-0000-4049-8000-36C40000603B               104.9 MB   disk6s3
   4:       Microsoft Basic Data                         7.5 GB     disk6s4
   5:       Microsoft Basic Data EASYROMS                275.8 MB   disk6s5

❯ diskutil mount /dev/disk6s5
Volume EASYROMS on /dev/disk6s5 mounted

❯ diskutil mount /dev/disk6s4
Volume on disk6s4 failed to mount
If you think the volume is supported but damaged, try the "readOnly" option

❯ diskutil mount /dev/disk6s3
Volume on disk6s3 failed to mount
If you think the volume is supported but damaged, try the "readOnly" option

❯ diskutil mount /dev/disk6s2
Volume on disk6s2 failed to mount
If you think the volume is supported but damaged, try the "readOnly" option

❯ diskutil mount /dev/disk6s1
Volume on disk6s1 failed to mount
If you think the volume is supported but damaged, try the "readOnly" option
```

Any hints? I don't have a Windows machine at my disposal right now.

christianhaitian, 2023-01-21:
That's odd, it should be easily accessible I thought even on a Mac since the boot partition is fat32.  It should be the 4th partition and is about 100MBs in total size.

scspijker, 2023-01-22:
Is looks like it has a GPT scheme, and 5 partitions, the 4.2MB ones are probably some kind of partition / FS descriptors, so that would leave a 104.9MB (boot?) partition, a 7.5GB (main?) partition and a 275.8MB partitions that will probably expand on first boot?
I'll try altering the partition table to FAT32 so maybe the mac understands it better.

scspijker, 2023-01-22:
Gdisk confirms that the partition table inside the `.img` file does not state that the 100mb partition is a fat32 partition, it has an unknown GUID code:

```
Command (? for help): p 
Disk ArkOS_RG353V_v2.0_01152023.img: 15601664 sectors, 7.4 GiB
Sector size (logical): 512 bytes
Disk identifier (GUID): 23000000-0000-4C4A-8000-699000005ABB
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 15601630
Partitions will be aligned on 2048-sector boundaries
Total free space is 96189 sectors (47.0 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1           16384           24575   4.0 MiB     FFFF  uboot
   2           24576           32767   4.0 MiB     FFFF  resource
   3           32768          237567   100.0 MiB   FFFF  ANBERNIC
   4          237568        14983167   7.0 GiB     0700  Basic data partition
   5        14983168        15521791   263.0 MiB   0700  Basic data partition

Command (? for help): i 3
Partition number (1-5): 3
Partition GUID code: 4F4C0000-0000-4049-8000-36C40000603B (Unknown)
Partition unique GUID: 223E0000-0000-4B12-8000-58CA00007B72
First sector: 32768 (at 16.0 MiB)
Last sector: 237567 (at 116.0 MiB)
Partition size: 204800 sectors (100.0 MiB)
Attribute flags: 0000000000000000
Partition name: 'ANBERNIC'

Command (? for help): i
Partition number (1-5): 4
Partition GUID code: EBD0A0A2-B9E5-4433-87C0-68B6B72699C7 (Microsoft basic data)
Partition unique GUID: 007800AD-96C0-4FD0-60EC-DB015A01F000
First sector: 237568 (at 116.0 MiB)
Last sector: 14983167 (at 7.1 GiB)
Partition size: 14745600 sectors (7.0 GiB)
Attribute flags: 0000000000000000
Partition name: 'Basic data partition'

Command (? for help): i
Partition number (1-5): 5
Partition GUID code: EBD0A0A2-B9E5-4433-87C0-68B6B72699C7 (Microsoft basic data)
Partition unique GUID: 007803AB-78A0-5045-60EC-FC015607F000
First sector: 14983168 (at 7.1 GiB)
Last sector: 15521791 (at 7.4 GiB)
Partition size: 538624 sectors (263.0 MiB)
Attribute flags: 0000000000000000
Partition name: 'Basic data partition'
```

So that explains why MacOS can't attach it, since the partition table does not reflect what type of filesystems are actually in there. The hardware device probably "expects" this setup and thus forces the kernel to read the partition as fat32.

scspijker, 2023-01-22:
@christianhaitian After changing the partition type, following your instructions on replacing `rk3566-OC.dtb` the devices boots succesfully 🎉 

What do the instructions actually do? I think replacing a kernel driver with a beta / stock driver? Maybe I can be of more assistance to this great project to increase support for "different" revision of the RG353V.

christianhaitian, 2023-01-22:
Wow.  This is terrible for Mac users.  Windows is able to read the fat partition without doing any of this.  Either way, not an ideal solution for any users to have to jump through these hoops.  The ultimate goal is to accommodate both screen types from the same dtb file like the stock os does.  The kernel we currently have access to is [here](https://github.com/christianhaitian/RG353VKernel).  If you have ideas on improving support for the v2 screens, let me know.

marcstyles, 2023-01-24:
@scspijker this worked for me as well. The device boots successfully however i now don't see the EASYROM partition when i load up the sd card. Any ideas how to reinstate this?

scspijker, 2023-01-26:
Thanks a bunch for the help!

I'll look into helping out with the kernel. I don't have experience in Linux kernel programming except for some drivers, so don't expect much though, haha.

closed 2023-01-26 by scspijker

christianhaitian, 2023-01-26:
Same experience ad me then. 😅

cblte, 2023-02-16:
This has helped me today. It took a couple of restarts to get the stuff right, but now it is working. I also activated the second SD Card. Hope I did it right by saying ROMS on TF2.

christianhaitian, 2023-03-31:
This solution is no longer needed for the RG353V and RG353M units with the updated images made available through the wiki.  Thanks to @dfreivald, @RetroGFX, and JelOS for the work on this.