christianhaitian/arkos #621: 01152023 RG503 PPSSPP issue

bell100812, 2023-01-19:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using? RG503

### What version of ArkOS are you on? 01152023
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

My RG503 has a issues when I use 01152021 ARKOS.
In the ppsspp, I can not use R1 R2 L1 L2 and analogue pad.
Please teach me how can I fix it.
before update to 0115 version, there are no issues like that.

Always I appreciate you.

Thanks and best regards.

christianhaitian, 2023-01-19:
Try deleting the ppsspp folder within the /roms/psp folder (or /roms2/psp if using 2 sd cards). If you have game saves you'd like to maintain, just delete the ppsspp/PSP/SYSTEM/controls.ini and ppsspp/PSP/SYSTEM/ppsspp.ini.sdl file and on the next launch, of the ppsspp standalone emulator, those files will be recreated that fix that issue

bell100812, 2023-01-20:
Hi.
I tried your advice.
I deleted controls.ini and ppsspp.ini.sdl.
As your advice, when I reboot the RG503, the files I deleted are restored.
But my problems still remained.
I can not use analogue sticks and R1, R2, L1, L2 buttons and the ppsspp
menu.
Is there another solution for that?
Thank you for your sincere support.

Thanks and regards

christianhaitian, 2023-01-20:
In `psp/ppsspp/PSP/SYSTEM/` try deleting controls.ini, ppsspp.ini, ppsspp.ini.go and ppsspp.ini.sdl and see if that fixes your issue.

christianhaitian, 2023-01-24:
Any update on this issue?  I haven't been able to replicate it on my end.

bell100812, 2023-01-24:
Thank you for your sincere reply.
But I can not fix my issues with your solution.

christianhaitian, 2023-01-24:
Then try a fresh image.

bell100812, 2023-01-24:
okay. I will try re-download the software.

closed 2023-01-24 by christianhaitian

bell100812, 2023-01-28:
I tried new image (01252023 ver) but the issues are still reamined

christianhaitian, 2023-01-28:
Are you copying your psp iso files from another ArkOS installation?  If so, don't copy the entire folder.  Just try copying 1 iso and see if it works without issue.