christianhaitian/arkos #622: RG353M | Resume from sleep mode fails

Gallocon, 2023-01-20:
**ArkOS Version:** 2.0 (01152023)

**Short version:** if you use sleep mode multiple times without a restart in between, the system will not wake up without a reset. 

**Expected behavior:** after the device and OS is powered on, when you tap the Power button. The device goes to sleep mode. When it's in Sleep, tap the Power button again to wake it up. This should be true regardless of the number of times you sleep and wake up the device. 

**Actual behavior:** After the device is powered on, pressing the power button puts the device to sleep. Then pressing power again will wake it up and resume where you left off. If you sleep again with the power button, the device will not wake up with another press of the power button there are no signs of life in this failed state (power indicator stays off, no audio from speakers, etc). The only way out of the failed state is to press Reset. The issue also can happen if you use Main Menu >> Quit >> Restart System between your activation of sleep mode attempts. The same does not appear to be the case with Shutdown System from the Quit menu. 

**Impact:**to those unknowing, this could cause loss of progress in games or settings for their cores they have not saved. 

**Workaround:** only use Sleep mode sparingly (which is undesirable). Save progress or state prior to sleep mode to prevent loss of data

This is not listed on the known issues page or on the FAQs for the 353M. 

As an end user ISBAT put the device to sleep and wake it up consistently without risk of losing progress.

If you have any questions about this report, please let me know!

christianhaitian, 2023-01-20:
I sleep my device multiple times between boots without issue.  The issues you're having seem to indicate a possible issue with either the sd card or the unit.  Try sleeping and waking your unit in Android and in the stock os multiple times and see if the issue happens there.

christianhaitian, 2023-01-20:
I should share that I sleep and resume my rg353m unit in various games and at different times throughout the day to due to my schedule and I never get a failed resume.  My current main OS card is a Samsung 32gb one but I've also done this on a rg503 with a Sandisk sd card without issue.  I also never let my unit fall below 70 percent charge which may impact reliability since the battery reading is not always accurate on these devices.  Low battery can cause unusual issues.  Lastly, I used quality authentic sd cards.  I don't use the cards that come with the unit and I tend to stick to new sd cards and not ones that have been used for use.  I've had a few used PNY cards fail on me.

Gallocon, 2023-01-20:
Thanks for the reply! 

My current SD card is a SanDisk Ultra 32GB. It's not the stock that came with the device. Android does sleep properly, however the Sleep is a bit different. On ArkOS, when the device is in Sleep, the power indicator turns off. In Android, the Power Indicator stays on. It may just be turning off the display (like when you press the lock button on your phone). 

Earlier this morning (when I was creating the OP), I was around 75% charge. Afterword, I charged fully while I was working, and while at 100%, I verified the issue still happens. This seems to happen every time regardless of core. Today my tests were on Legend of Zelda: Oracle of Seasons (using Retroarch - Gambette) and Celeste Classic (on Pico-8 - Auto). 

FWIW, [there have been others reporting similar issues](https://www.reddit.com/r/ANBERNIC/comments/zcr3x4/rg353m_problem_with_sleep_mode/). It's not a lot, but they exists (there could be DOZENs of us!) 

I'll give stock firmware a try and see if the issue persists, if not, I'll do a fresh install on a different (quality) SD card too. When I do I'll report back. In the meantime, is there any kind of function that you know if that Sleep relies on? Like a form of cleanup that has to run from it's past sleep states?

christianhaitian, 2023-01-20:
There are functions that sleep does to enter sleep but they wanted affect the unit from waking.  At that point, it's between the kernel and the device to wake the unit from it's sleep state.  Not much I can do to fix that unless I can replicate the issue which I can not since sleep works perfect for me.  The few other reports I've gotten on this issue usually ended up being an issue with the device itself that people have been able to resolve by getting their units replaced.

Gallocon, 2023-01-20:
I tested with JELOS (from before they stopped support), and it seems the issue persists there too. So I took that same SD card, and installed Stock firmware. And plot twist!: the issue doesn't happen! 

I'm going through and cleaning and reinstalling ArkOS from scratch. I'll report back my findings!

Gallocon, 2023-01-20:
And the verdict is: issue still exists. This was both imaging to the existing SD card, and the known good one that was used for Stock OS. 

I've shot an email over to Anbernic to see if they can do anything. Wish me luck!

JoeLyga, 2023-01-21:
I'm having the same issue. Sometimes when I press the power button to put the device to sleep, I can't get the device to wake up unless I reset or eject the system SD card. 

I'm also wondering if it's related at all to power settings under advanced settings. Ive tried various states (instant, default, etc) but am not sure what it effects. 

Thanks!

Gallocon, 2023-01-22:
I'm wondering if there is any value in enabling debug logging and then compare with a working unit's logs. If Sleep is working on the stock OS, maybe there could be a tweak to how ArkOS handles the sleep in those cases.

christianhaitian, 2023-01-22:
The fact that your unit doesn’t wake up reliably from a sleep with cfw would make this difficult without a serial cable.  Out of curiosity, have you tried turning off Bluetooth and seeing if that helps?

Gallocon, 2023-01-22:
Yep! I've turned off Bluetooth for all OS versions I've tried. No need to have it on if it's not in use. Might as well save the battery.

christianhaitian, 2023-01-22:
How about Wifi?  You can turn that off in Options>Advanced>wifioff

Gallocon, 2023-01-22:
Same issue with Wi-Fi off. 

I disabled Wi-Fi, completed a shutdown and power on. Then confirmed in the main menu that Wi-Fi was Off (Bluetooth was still off as well)

Opened a game, and tapped power for sleep mode. Then woke up the device. Played for a few moments, then tapped power again. Sleep started, but the device would not wake or show signed of power. 

And while I was at it, I checked Main Menu >> Advanced Settings >> Power Save mode, which was set to DISABLED.

christianhaitian, 2023-01-22:
Does the issue happen on the same emulator or game or is it with different emulators and games as well?

Gallocon, 2023-01-23:
It happens regardless of core/emulator. Tested with RetroArch Gambette for Oracle of Seasons, as well as Ascent on Pico-8 (official install, not Fake-08)

christianhaitian, 2023-01-23:
It will be an interesting experiment.  If possible, also try testing with a different sd card brand like Samsung.  On my end, I get consistent results with both Sandisk and Samsung.

Martyred13, 2023-01-24:
Made an account just to comment that I also have an issue with getting sleep to work correctly. 

Device: Anbernic RG353M (Purple)
OS Running: ArkOS_RG353M_v2.0_01152023
SD Card: SanDisk 18GB Ultra (Also tried SanDisk 64GB Ultra)

Steps to reproduce from home screen:
ArkOS Home Screen
Short press power button (Device goes into sleep)
Short press power button (Device wakes)
Short press power button no longer has any affect on the device. Can't put device in sleep.

Steps to reproduce from inside emulator:
Launch game 
Short press power button (device goes into sleep)
Short press power button no longer has any affect on the device. Can't wake the device
Only option to get the device to wake up at this point is to press the "R" button to restart ArkOS

Gallocon, 2023-01-24:
Looks like that second Rg353M I ordered should be delivered today. So assuming I'm not getting screwed and getting shipped a fake, I should be able to test further later today! I do have available Samsung EVO cards I can use for testing too (though the storage on them is overkill). 

I'll keep y'all posted!

christianhaitian, 2023-01-24:
When a device is awaken, it takes about another 15 seconds before the device can be put to sleep again as the daemon that watches for global hotkey input is restarted.  However, waking from sleep is controlled by the hardware at that point.  I wonder if Anbernic's stock OS doesn't fully put the unit to sleep probably because newer devices have an issue.  I've seen a few comments that sleep in Linux on stock seems to drain the battery more than cfw which would further support the theory.

Gallocon, 2023-01-24:
That would totally track. I'd be curious why it's split and inconsistent though. If your hunch is correct, wouldn't you expect to be able to duplicate the issue? Unless there are different hardware variants or this is a silicon lottery thing.

christianhaitian, 2023-01-24:
Could be there were some changes to newer mfg devices that are being handled differently.  I wouldn't be surprised as again, these are cheap chinese devices at the end of the day.  Quality control is not a priority like it would be for established major brands like a Nintendo, Microsoft or Sony.

christianhaitian, 2023-01-24:
[Image.zip](https://github.com/christianhaitian/arkos/files/10492899/Image.zip)
I'm flying blind here as I can't reproduce these issues on the my end as sleep works as expected repeatedly without issue.  I've updated the kernel to enable some additional sleep functions.  I haven't noticed any ill effects on my end during testing.  Others are welcome to test and report if it helps with stability of sleep.  It can be used on RG353V/VS and RG353M units.  Just rename the existing `Image` file in the boot partition to something like `Image.bak` and copy this one into the boot partition.  Boot your device back up and test sleep and see if there's a difference in stability.

Martyred13, 2023-01-24:
Downloaded the image file you shared and tested it out really quickly.

New behavior:
Game running short press power button device goes to sleep
Short press power button device wakes up
Wait about 10 seconds
Short press power button device goes to sleep
Device can no longer be awaken with short press only a restart of the OS works.

This is the same behavior on the OS menu screen as well.

christianhaitian, 2023-01-24:
Did you make sure to wait until the power light goes out to confirm the device goes to sleep?

Martyred13, 2023-01-24:
I retested waiting longer time intervals (39-60 seconds) and same result as above.

rolivav, 2023-01-24:
I am seeing the same issue on a RG353VS. I do not know when it appeared, I've only had this device for a week so I already had the latest version.
It is exactly as described on the ticket, only I want to add that sometimes if you send it to sleep and then resume (if it works) the power button does not send it to sleep anymore, you can hit it as much as you want that it will not go to sleep.

christianhaitian, 2023-01-24:
[Image.zip](https://github.com/christianhaitian/arkos/files/10493307/Image.zip)
@Martyred13 Can you test this one and see what results you get?  Here I disable most suspend states available in the kernel and depend on just the OS.

Martyred13, 2023-01-24:
Same results, works for one sleep / wake cycle and then after you put it to sleep it no longer functions anymore. Power button does nothing and you have to press the reset button. Is there any way with ARKOS to print you a log file?

christianhaitian, 2023-01-24:
Not if the device is rebooted. :(

christianhaitian, 2023-01-24:
Any relevant data would be gone

rolivav, 2023-01-24:
any way to persist that data somehow so we can help?

Martyred13, 2023-01-24:
I tried the steps with the logging set to debug and just not doing the last sleep press that I know locks the device but no log file was generated. It would be in /tmp/logs right?

christianhaitian, 2023-01-24:
I'm reviewing the processes that the stock os does and it looks like it sets a particular state to `/sys/power/state` before it goes to sleep.  If you're comfortable with ssh, run the following from within a ssh terminal session, then immediately test sleep to see if anything changes.

`sudo sed -i "/#SuspendState\=/c\SuspendState\=mem" /etc/systemd/sleep.conf`

Gallocon, 2023-01-24:
I've given the command a try via Putty. No output, which appears to be normal. I rebooted and tested, I'm still having issues where after the second sleep is started, the device will not wake. 

As for that Image.zip file, I do not have an active Linux machine move the file. I've tried using SFTP, WSL to mount the drive, and using Putty pscp for the transfer. SFTP seems to have permissions errors writing to /boot/. WSL and pscp feel like fool's errands. lol. Any pointers for Windows plebs to get that Image copied over?

christianhaitian, 2023-01-24:
Check that /etc/systemd/sleep.conf has a line that says `SuspendState=mem`

Gallocon, 2023-01-24:
Yep! `SuspendState=mem` is present. However, it's the only option on the [Sleep] table that is not commented out with `#`. I'm uncertain if `#` would normally be to signal a comment in conf files though.

christianhaitian, 2023-01-24:
The `#` in front of the rest of the options is find.  It just allows the kernel to set those values by default if needed.  Try the following via ssh and then test sleep and see if your results are any different. `sudo sed -i "/#SuspendState\=/c\SuspendState\=freeze" /etc/systemd/sleep.conf`

christianhaitian, 2023-01-24:
also, I noticed that the sleep script located at `/usr/lib/systemd/system-sleep/sleep` has some errors that need to be fixed.  You can do a `sudo nano /usr/lib/systemd/system-sleep/sleep` and replace the contents in there with the following.  I doubt it helps with the wake issue but who knows.

```
#!/bin/bash

case $1 in
   pre)
    test=$(top -b -n 1 -d 0.2 -p $(pidof emulationstation) | tail -1 | awk '{print $9}')
    if (( ${test%.*} > 100 )); then
      systemctl stop emulationstation
    fi
    echo $(cat /sys/class/backlight/backlight/brightness) | sudo tee /var/local/cur_bright.state
    echo 0 | sudo tee /sys/class/backlight/backlight/brightness
    sudo /usr/sbin/alsactl store -f /var/local/asound.state
    sudo btconnected.sh check
    ;;
   post)
    echo $(cat /var/local/cur_bright.state) | sudo tee /sys/class/backlight/backlight/brightness
    sudo /usr/sbin/alsactl restore -f /var/local/asound.state
    systemctl is-active --quiet emulationstation.service && echo ok || systemctl start emulationstation
    sleep 2
    #sudo systemctl restart oga_events
    systemctl is-active --quiet enable_bluetooth.service && sudo systemctl restart enable_bluetooth
    systemctl is-active --quiet bluetooth.service && sudo systemctl restart bluetooth
    systemctl is-active --quiet bluealsa.service && sudo systemctl restart bluealsa
    systemctl is-active --quiet watchforbtaudio.service && sudo systemctl restart watchforbtaudio --no-block
    if [ -f "/var/local/btautoreconnect.state" ]; then
      sleep 15
      sudo btconnected.sh reconnect
      sudo rm -f /var/local/btautoreconnect.state
    fi
    systemctl is-active --quiet solarushotkey.service && sudo systemctl restart solarushotkey
    systemctl is-active --quiet pico8hotkey.service && sudo systemctl restart pico8hotkey
    systemctl is-active --quiet ppsspphotkey.service && sudo systemctl restart ppsspphotkey
    systemctl is-active --quiet openborhotkey.service && sudo systemctl restart openborhotkey
    systemctl is-active --quiet singehotkey.service && sudo systemctl restart singehotkey
    ;;
esac
```

Gallocon, 2023-01-24:
I'll keep that in my back pocket, but GOOD NEWS!: Setting `SuspendState=freeze` is actually sleeping and waking normally and without issues!!!

christianhaitian, 2023-01-24:
woah!

christianhaitian, 2023-01-24:
Please try copying the `Image.bak` file back to `Image` and see if it continues to work there as well.

Gallocon, 2023-01-24:
I never got to copy the Image.zip (due to complications being able to copy the Image file from a Windows machine. the only tweaks from a base install we have right now is:

Bluetooth is disabled
SuspendState=freeze

Other than that, this is a stock image from the latest release.

Martyred13, 2023-01-24:
I just dragged the Image file into the boot folder for the SD card on windows. Is that not the correct way?

Gallocon, 2023-01-24:
You might have had better luck than I had getting the Linux partitions to show up in File Explorer. It's....messy. Unless there is an easier way to do that that I've forgotten about or am oblivious to (which is totally possible lol)

Martyred13, 2023-01-24:
Oh, it's still messy. Like 50 windows open up and I have to cancel a request to format about 50 times, but it works haha

Gallocon, 2023-01-24:
I'm an idiot and overcomplicated it. Sorry all! OTL

christianhaitian, 2023-01-24:
Either way, good to know that changing that one setting to freeze seems to help with the stability.  I haven't had any issues with it on my end either so hopefully this should improve the situation for most others.

Gallocon, 2023-01-24:
Thanks for dealing with all of these things and helping! We appreciate everything you do!

Martyred13, 2023-01-24:
@christianhaitian I went back to the original Image file and if I set the `sudo sed -i "/#SuspendState\=/c\SuspendState\=freeze" /etc/systemd/sleep.conf` before I hit the short press on the power button it comes back from sleeping every time!

christianhaitian, 2023-01-24:
Thanks everyone for the feedback and testing.  I'll get an update out asap for this.

closed 2023-01-24 by christianhaitian

christianhaitian, 2023-01-27:
Great to hear!  

As for an additional device.  Not at this time.  Thanks for the offer though.