christianhaitian/arkos #623: Issue with WiFi networks and spaces (parsing?)

JoeLyga, 2023-01-21:
RG353M, ArkOS 01152023. 

My WiFi network had a space in it.  It's "J & L" without the quotes. Sometimes when I power off and start the firmware, I've noticed that my WiFi connection isn't working (this may be after a hard reset). When going into the WiFi settings, I notice that my WiFi network is spread across 3 lines with 3 separate entries. 

J
&
L

Activating any of the 3 lines doesn't work. But re-scanning and adding the J & L network does work. Seems like there's some parsing issue with spaces perhaps.

christianhaitian, 2023-01-21:
Yes, I can recreate the issue.  It is a parsing issue but I have no idea how to solve that issue yet other than to not include spaces and special characters in SSIDs in general.  See this [link](https://routersecurity.org/SSID.php) on why special characters and spaces are not a good idea in SSIDs.  They may work fine in some circumstances but can introduce weird issues in others.

closed 2023-01-21 by christianhaitian

JuanMiguelBG, 2023-01-24:
To manager SSIDs withd spaces, an example method to return a list of SSIDs

`do_list() {
    #print_log $LOG_FILE "INFO" "Executing 'do_list()'"

    local networks=$(sudo nmcli -f ALL --mode tabular --terse --fields SSID dev wifi | sort -u)
    local result=""
    while IFS= read -r line; do
      #print_log $LOG_FILE "DEBUG" "Executing 'do_list()' - line: '$line'"
      local net_regexp=`echo "$line" | tr -s " "`
      net_regexp=${net_regexp=// /\s}
      #print_log $LOG_FILE "DEBUG" "Executing 'do_list()' - net_regexp: '$net_regexp'"
      local exist=$(sudo ls -1 /etc/NetworkManager/system-connections/ | egrep "$net_regexp")
      #print_log $LOG_FILE "DEBUG" "Executing 'do_list()' - exist: '$exist'"
      # skip empty values
      local line_aux=$(trim "$line")
      #print_log $LOG_FILE "DEBUG" "Executing 'do_list()' - line_aux: '$line_aux'"
      if [ ! -z "$line_aux" ]; then
        if [ ! -z "$exist" ]; then
          result="$result\n$line_aux$EXIST_CONNECTION_FLAG"
        else
          result="$result\n$line_aux"
        fi
    fi
    done <<< "$networks"
    if [ -z "$result" ]; then
      echo ""
    else
      echo -e "${result#\\n}"
    fi

    #print_log $LOG_FILE "DEBUG" "Executing 'do_list()' - result: '$result'"
}
`