christianhaitian/arkos #625: Play a video via command line?

benckx, 2023-01-23:
I'd like to be able to play a video in a loop at the start-up of the system (it's for an art exhibition, I'd like to make it so the device can be rebooted and the video just starts playing with no hassle).

So I'm looking for a command I could add at the end of the `.bashrc`.

I'm able to connect in ssh and play the video with `ffplay`, but I can only hear the sound and the screen still shows the OS GUI.

I noticed if File Manager is running, then I can see the video when I run `ffplay` via command line. So I tried running the File Manager with command `/opt/dingux/DinguxCommander` before playing the video. But again, File Manager doesn't show on display.

It's like something prevents from getting "priority" to the display somehow?

Workarounds I found:
- Wait for the screensaver before playing the video. But it would take 5 min. Is there a setting to reduce that time?
- Run command `systemctl stop emulationstation.service`, then run `ffplay`. But I'd need sudo.

I guess there must be a simpler way?

The device is a RG351M.

christianhaitian, 2023-01-23:
Why not just disable emulationstation and just have your video start after boot instead?  You can disable emulationstation by doing `sudo systemctl disable emulationstation` then create a new systemd that just plays your video or just add it to the end of the `/home/ark/.bashrc` file.

benckx, 2023-01-23:
I guess I could try that.

If something is misconfigured after I disabled `emulationstation` from booting, can I easily re-enable "network service"? I could probably also enable ssh on start-up.

christianhaitian, 2023-01-23:
Just do `sudo systemctl enable ssh` and ssh will always be available

christianhaitian, 2023-01-24:
Good luck and enjoy.

closed 2023-01-24 by christianhaitian