christianhaitian/arkos #626: Bluetooth audio not working well

rolivav, 2023-01-24:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Bluetooth headphone issues! I have two pairs of bluetooth headphones.
- Samsung Galaxy Buds Live - they connect well, but no sounds comes out of them. Restarting emulation station and making sure that the "headphone mode" is set, nothing.
- Urbanista Lisbon - these ones do have sound, but there is about 1 second of latency.

### How can the issue be reproduced? (If applicable)
I guess not easy to reproduce for you if you don't have these devices :/

### What device are you using?
RG353VS

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
01152023

### Anything else you'd like to include that may help to review this issue or feature request?
Let me know what I can help with!! I am fairly new to this scene so I am completely clueless but if you need logs or anything else just ask.

christianhaitian, 2023-01-24:
Not sure how best to help on this.  latency has been reduced as best as possible and per this [issue](https://github.com/christianhaitian/arkos/issues/612), the latency seemed to have improved nicely for that person.  It probably depends on the headset.  I have an old Philips bluetooth headset that seems to have ok latency.  I also have a Anker Soundcore bluetooth headset I purchased cheap on Amazon that seems to perform well with very minimal latency.  If you're familiar with ssh, you can try to edit the /home/ark/.asoundrcbt file and change the period and buffer size to see if there's a number that works better for you there.  Good luck.

closed 2023-01-24 by christianhaitian

rolivav, 2023-01-24:
Anyway thanks for the tips, I'll SSH into the device and tweak those values. I assume I have to reboot the device everytime I change them, or can I just restart the bluetooth?

rolivav, 2023-01-24:
@christianhaitian you might not be getting any more notifications for this topic, but I'll be digging a bit on this.
I also tried a pair of Razer Kraken BT and they work the same way as the Urbanista Lisbons. So I currently have no headphones that work perfectly well.

christianhaitian, 2023-01-24:
I can still see updates to closed tickets.  A quick way to not have to reboot between tests is to just edit the `/home/ark/.asoundrc` file instead while bluetooth is connected and see if different periods and buffers improve there.  

btw, where are you noticing the latency issues the most?  If within a retroarch emulator, try setting the retroarch audio latency to 0 and see if that helps. Settings>Audio>Output>Audio Latency (ms)

rolivav, 2023-01-24:
That's a good point, I didn't mention I notice this in the gba core. I tested setting the audio latency to 0 and funny enough, it seemed to make it worse!

rolivav, 2023-01-24:
I also get the latency on the emulation station menu, so I think this happens everywhere 🤔.

rolivav, 2023-01-24:
Mmm I was changing this setting from the quick menu but I am not sure it works at all there. Disabling the audio in that menu doesn't disable it in game.
I am trying to change the setting from retroarch in the main menu, I load the mGBA core but after changing the setting, it doesn't seem to persist. I think I am missing basic knowledge of how to set things in Retroarch.

christianhaitian, 2023-01-24:
save configuration

rolivav, 2023-01-24:
Big facepalm moment, thanks. That didn't do anything either. It is still almost a full second of latency...
Looking into SSHing into it now.