christianhaitian/arkos #627: Menus contain a mix of English and another language

wilsonam, 2023-01-26:
### Describe the Issue (If applicable)
Menus accessed by pressing START in the main interface contain a mix of English and another language (Portuguese I think).

### How can the issue be reproduced? (If applicable)
From the main UI where you select which system you want to go into, press START. Notice that some menu items are in English and some are not. This issue also exists when you press START inside a list of games for a specific system.

### What device are you using?
RG353M

### What version of ArkOS are you on?
2.0 (01252023)

### Anything else you'd like to include that may help to review this issue or feature request?
I updated my device over wifi on 1/25/2023. This issue did not exist before I performed this update. I have not tried to do a fresh install of ArkOS by writing it directly onto my microSD.

christianhaitian, 2023-01-26:
I can't replicate this issue on my end.  Did you change languages in the start menu>Advanced Settings>Language?

wilsonam, 2023-01-26:
I did not. Thanks for giving me the path to the language settings. The language was set to `BR`. I switched it back to `EN` and that resolved the issue. I tried switching back to `BR`, and that worked as expected too. There are still a couple items that are in English (I assume that is the fallback when a translation string is missing), but before it was about 50% English 50% Portuguese. I'm going to chalk it up to an anomaly or user error on my part. Thanks for the support!

closed 2023-01-26 by wilsonam