christianhaitian/arkos #629: RG353VS（01252023） can't find and pair up bluetooth 5.2 device

JackyWWX, 2023-01-31:
After update to 01252023，RG353VS can't find and pair up bluetooth 5.2 device，such as TWS device（onplus budspro 2），I have try many times and turn on/off bluetooth，or repair tws device，can‘t be solved.but rollback to 01152023，it works fine.
also，bluetooth 5.0 TWS （SONY wf1000XM4）may works fine on 01252023.

christianhaitian, 2023-01-31:
There were no changes that impacted bluetooth pairing between 01152023 and 01252023.    Even if there was, I can not gaurantee that all bluetooth headphones will work since I don't own many of them.  Only advice I have  is if you don't see your bluetooth device show up after verifying you put your head phones in pairing mode, try the search more than once.

closed 2023-01-31 by christianhaitian

JackyWWX, 2023-01-31:
but I try to rollback 0152023，repair TWS，it's work correctly.And reupdate to 01252023，active exsting bluetooth，then the device name is disappear. once again，can't find and pair up. This is annoying and confusing