christianhaitian/arkos #630: Pokemon Mini 7z Support

nicjansma, 2023-02-03:
### Describe the Issue (If applicable)
[Pokemon Mini](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#pokemon-mini) through RetroArch should have 7zip support.  In the [Wiki](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#pokemon-mini) it only lists Zip support and not 7z, and if I only include 7z files in `pokemonmini/` dir, the system does not show up.

EmulationStation scanning should be updated to include `.7z .7Z` files.

### How can the issue be reproduced? (If applicable)
Having a `pokemonmini` directory full of `.7z` files doesn't show up in ES.

Having a `pokemonmini` directory full of `.zip` files shows up and can be played.

On other platforms (e.g. SteamDeck/RetroDeck or OnionOS) 7z is supported via RetroArch for pokemon mini.

### What device are you using?

RG353V

### What version of ArkOS are you on?

2.0 (01032023)

christianhaitian, 2023-02-03:
Ok.  I’ll add this in the next update.

christianhaitian, 2023-02-03:
Also, your version of ArkOS is old.  You should update to the latest 01152023 to get the latest stability fixes especially with sleep.  Just do Update from the Options section while connected to wifi.

closed 2023-02-10 by christianhaitian