christianhaitian/arkos #631: RG353VS NDS emu on HDMI out，monitor can‘t find HDMI signal

JackyWWX, 2023-02-03:
I try RG353VS  output via HDMI，in ES menu and other emu，Normal play。but when I try NDS emu，the game screen is black，monitor can‘t find HDMI signal。Then press L3 into emu menu，the signal can be found，but quite the emu menu then break again.

christianhaitian, 2023-02-04:
I was able to confirm this issue.  However, this is not something that can be addressed at this time or ever since that emulator (Drastic) is closed source.

christianhaitian, 2023-02-04:
I was able to find a fix for the issue.  I will include it in the next update.  It will also fix PSP.

JackyWWX, 2023-02-06:
> I was able to find a fix for the issue. I will include it in the next update. It will also fix PSP.

great！wait for update

christianhaitian, 2023-02-10:
New update has been released that should resolve this issue.

closed 2023-02-10 by christianhaitian

JackyWWX, 2023-02-13:
> New update has been released that should resolve this issue.

awesom!The HDMI issue and Bluetooth audio issue have all been solved. I have been playing normally for 3 days.THX for fix.