christianhaitian/arkos #632: Ability to disable led for TP-Link TL-WN725N on 351MP

CrazyXacker, 2023-02-04:
### Describe the Issue (If applicable)
Is there any way to disable led for TP-Link TL-WN725N wifi module? I have strange issue with it: when led blinking, speaker is crackling. It's very annoyning. On AmbelELEC this led is somehow disabled by default and there is not any crackling at all

### What device are you using?
Powkiddy RGB20S (351MP clone). ArkOS is working fine on this device

### What version of ArkOS are you on?
Latest for 351MP from Wifi update

christianhaitian, 2023-02-04:
I don’t know.  I don’t support the rgb20s.

closed 2023-02-04 by christianhaitian

CrazyXacker, 2023-02-04:
@christianhaitian i'm just asking for ability to disable wifi led. it's not "platform" related request as i think

christianhaitian, 2023-02-04:
Not that I'm aware of.  It may be possible but it's not something I've investigated.