christianhaitian/arkos #633: Shoulders buttons still not working in some NAOMI games

zhuwz1, 2023-02-04:
Try removing the -triggers option.

_Originally posted by @christianhaitian in https://github.com/christianhaitian/arkos/issues/368#issuecomment-834709211_

Hi, I am using ArkOS 2.0 in RG351MP. I have already removed "-n" and "-triggers" in naomi.sh. Now the analog stick works, but the shoulder buttons still don't work in games like CVS and CVS2. Is there a way to fix this? Thanks.

christianhaitian, 2023-02-06:
I'm assuming you're using RertroRun?  If so, do they work with Dreamcast games using Retrorun?

zhuwz1, 2023-02-07:
Yes. I am using RetroRun32 with FLYCAST32_RUMBLE core. In Dreamcast games, the shoulder buttons L2 and R2 are effective by default. But in NAOMI games, no shoulder button works even after I delete "-triggers" in naomi.sh. 

Also in Dreamcast games, is it possible to use L1 and R1, or use all 4 shoulder buttons together? I mean mapping both L1 and L2 buttons as L trigger of DC controller.

RetroGFX, 2023-02-11:
> Yes. I am using RetroRun32 with FLYCAST32_RUMBLE core. In Dreamcast games, the shoulder buttons L2 and R2 are effective by default. But in NAOMI games, no shoulder button works even after I delete [...]

this should be possible.. it would just point both triggers to the same command i would think

christianhaitian, 2023-02-18:
The L2 and R2 buttons not functioning as triggers has been addressed with today's update.  No solution for L1 and R1 not working with Sega emulators yet.  If a solution is found in the future, it will be included in a future update.

closed 2023-02-18 by christianhaitian