christianhaitian/arkos #634: RG353VS（0115-0125） bluetooth audio will suddenly disappear in NDS emu

JackyWWX, 2023-02-07:
RG353VS test on V 0115-0125：
In NDS emu, the Bluetooth audio is normal when just enter the emu and openup a game, but it will suddenly disappear when you play, but the Bluetooth status is still connected. The specific time of disappearance is uncertain, and the cause is unclear. This is the case for different tws or speakers.
once the Bluetooth audio is normal during a long-time game for over 1h, but the delay gradually increases.
also the built-in speaker does not have this issue.

JackyWWX, 2023-02-08:
After troubleshooting and testing the options of the NDS emu one by one, it was found that as  the high-resolution 3D option was turned YES, it would cause the issue.

christianhaitian, 2023-02-08:
If that is the case, I’ll try setting performance mode for drastic in the next update to see if that helps.  If not, not sure what else can help with that issue other than just not setting high res 3d when using bluetooth audio on this chipset.

JackyWWX, 2023-02-08:
> If that is the case, I’ll try setting performance mode for drastic in the next update to see if that helps. If not, not sure what else can help with that issue other than just not setting high res 3d when using bluetooth audio on this chipset.

OK，THX，wait for try it

christianhaitian, 2023-02-10:
New update has been released.  Hopefully it helps with this issue.

closed 2023-02-10 by christianhaitian