christianhaitian/arkos #635: Bluetooth audio device disconnects shortly after sleep and resume

trr, 2023-02-07:
Latest arkos (25 Jan 2023) on RG353VS

### Describe the Issue (If applicable)

When sleeping and resuming, bluetooth audio stops moments after resuming.  Sound is played for a short while (1 second or so) before it stops, and it turns out that the bluetooth device was disconnected and needs to be "activated" again in the bluetooth menu.

### How can the issue be reproduced? (If applicable)

Connect to bluetooth audio. 
Start a game (I was in a mega drive game, not sure if relevant).  
Sleep the device.  
Wake the device.
Observe that the audio resumes momentarily and then cuts out.

Then go into the bluetooth menu and you will be able to confirm that the bluetooth device is not connected anymore.

### What device are you using?

RG353VS

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

25 Jan 2023 (after online update)

### Anything else you'd like to include that may help to review this issue or feature request?

Audio device is an Edifier NB2 Pro, it's just a set of bluetooth earbuds.

The amount of time it was asleep for doesn't seem to make a difference, even sleeping for 30 seconds does it.

It is particularly interesting that the headphones did not disconnect while asleep, because sound resumes through the headphones for a little while after waking, before then disconnecting.

christianhaitian, 2023-02-08:
Hi,

Be aware that reconnection upon wake can take up to 30 seconds to complete which could include a disconnect and reconnect during that time, however, per the changelog from 12/23/2022, a workaround for auto reconnecting connected bluetooth devices upon wake from sleep was added but may not always be reliable.  It's been reliable for me using a philips and anker bluetooth device, but no guarantee it will be consistent.   Reliability will hopefully improve overtime with future updates of bluetooth audio when they occur.

Thanks,

closed 2023-02-08 by christianhaitian

trr, 2023-02-08:
Just tried and can confirm it did disconnect and then reconnect some time later (maybe 30 seconds).

I'm skeptical that the disconnect is needed as it seems to work for that short time after waking before it disconnects - I assume the disconnect/reconnect was done for compatibility with some kind of misbehaving devices

christianhaitian, 2023-02-08:
That’s correct.  My Philips BT headphone would reconnect upon wake without issue but my anker unit would not.  I also got reports from others with the same occurrences.