christianhaitian/arkos #638: RG353V Turns on to half chopped white screen

bcantrell427, 2023-02-10:
I have 3 RG353's in front of me. I have 3 identical Micro SD cards. 2 of the units boot from any of the 3 SD cards but one will not at all. It does go through the steps of booting with nothing displayed but when it is finished  all I have is a chopped off white screen that eventually turns blue. No text at all is displayed and there is a half inch wide black bar down the side on the left. It will boot from the OE SD card without issue and is fully playable.  it also works in android mode.  also  it never shows the ARKOS splash screen.

RetroGFX, 2023-02-10:
Please see here
https://github.com/christianhaitian/arkos/issues/620

christianhaitian, 2023-02-10:
Yea sounds like you have a unit with the updated screen.  As RetroGFX posted above, see that issue for how to workaround that issue for now.

christianhaitian, 2023-02-10:
It was pinned at the top of the issues page.  Enjoy!

closed 2023-02-10 by christianhaitian