christianhaitian/arkos #639: Feature: request for kodi plugins iptv client

yzzyzz, 2023-02-12:
the rg353 series surport kodi media player.
but the arkos didn't complie iptv client include. and the kodi official dosent support install by yourself due to dependence error.

hopefully the arkos will support  iptv client inbox like the official system .
the plugins addr: https://github.com/kodi-pvr/pvr.iptvsimple

christianhaitian, 2023-02-12:
Odd, not sure why that pvr plugin wasn't built and added.  I should have a solution available to test and confirm functionality shortly.

christianhaitian, 2023-02-12:
Hi,

I just built this plugin and Kodi seems to have loaded it fine. If you can help test this and provide feedback on if it functions as expected or not, I'd appreciate it.  

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, launch Kodi, configure the iptv client and verify functionality.

Let me know how it works out for you.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10716288/arkosbackup.tar.gz)

yzzyzz, 2023-02-12:
yeah,i've tested and it it works !!
my device is rg353vs v2. the latest arkos version:ArkOS_RG353V_v2.0_01152023
thx for your great work.

christianhaitian, 2023-02-12:
Awesome.  Thanks for confirming.  I'll include it with the next update as well for others.

closed 2023-02-12 by christianhaitian