christianhaitian/arkos #640: GBA and retroarch

Matroc21, 2023-02-14:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-02-14:
First I'm hearing of this issue.  I can't replicate it on my end either.  what sd card are you using and what program did you use to image ArkOS onto the sd card?

Matroc21, 2023-02-14:
win32 and im using a 32 gb lexar card to house the os and i enabled the setting on ARKos to use the secondary sd card slot, that one is a samsung 256gb

christianhaitian, 2023-02-15:
Did you fully charge the device?  If so, do you have Sandisk or Samsung card you can use as the main os card?  I don’t know how well Lexar cards will work.

christianhaitian, 2023-02-16:
Any update on this?

Matroc21, 2023-02-17:
yes so ive found out it is only with certain roms
and they are pokemon rom hacks to be exact
Pokemon R.O.W.E and Pokemon Radical red they worked and saved fine on stock and black seraph android they dont save normally and the save state will crash the system if i try to quit retroarch

cblte, 2023-02-17:
Good morning. Maybe your roms are a bit broken. ROM hacks are  a bit fragile. Create save states a lot to make sure you saved. Maybe try to get a new version of the ROM. There is a big archive.org archive with a lot of pkm rom hacks.

christianhaitian, 2023-02-17:
The gba emulator core in ArkOS is more up to date than what’s available in the stock os.  You can try checking to see if there’s a newer MGBA emulator core available that may solve the compatibility issue by going into retroarch, settings, core updater and doing an update there to see if the issue resolves.  If not, try another emulator core like gpsp to see if that works.  Otherwise, not more I can help with on this specific edge case.

Take care

closed 2023-02-17 by christianhaitian

Matroc21, 2023-02-18:
Hopefully this Is of anyuse