christianhaitian/arkos #641: Resume from Sleep mode issue - RG351V

Dantasstic, 2023-02-16:
### Describe the Issue (If applicable)
Resume from sleep initiates a refresh of Emulation Station. This did not happen on the previous build of ArkOS for this device.

### How can the issue be reproduced? (If applicable)
Press the power button while at the main menu to sleep the device, then tap it again to wake it up.

Video of the issue occurring: https://streamable.com/brrjjx

### What device are you using?
RG351V

### What version of ArkOS are you on?
ArkOS (02092023)

christianhaitian, 2023-02-16:
Yes, I see the issue on my end as well.  Must be related to the new sdl from the last update.  Must not like seeing the controller disconnected then reconnected

christianhaitian, 2023-02-16:
I'll provide a workaround for the issue with the next update.

Dantasstic, 2023-02-17:
Sounds good! 

Will the next update also bring RA 1.14.0 to the 351V? I noticed it is still on 1.13.0, while my RK2020 has 1.14.0.

christianhaitian, 2023-02-17:
Yes or newer if there is a newer retroarch available by then.  Not sure why it didn’t get into the 351v update some time ago.

christianhaitian, 2023-02-18:
This fix has been pushed along with the retroarch 1.14 update.

closed 2023-02-18 by christianhaitian

Dantasstic, 2023-02-18:
Oh wow, that was fast! I was thinking it would be on the next update cycle for all the devices! I appreciate all your hard work! 👍👍