christianhaitian/arkos #642: PS4 controller can cause ES automatic reboot on 353vs 02092023

JackyWWX, 2023-02-17:
I use rg353VS and PS4 controller via bluetooth，when update to 02092023 . On ES main UI ，ss soon as the PS4 controller is connected via  bluetooth ,  the ES will automatically reboot , and the indicator light of the PS4 controller will turn off.
If playing game , connect the PS4 controller , 353vs won't reboot , but the indicator light of the PS4 controller will turn on than off，as deactived .

christianhaitian, 2023-02-17:
Does the entire system reboot or is it just emulationstation that crashes and restarts?

christianhaitian, 2023-02-17:
Nevermind, i see the issue happening with my 8bitdo Lite2 bluetooth controller as well.

christianhaitian, 2023-02-17:
Fix will be available shortly

christianhaitian, 2023-02-18:
Fix has been pushed with today's update.

closed 2023-02-18 by christianhaitian

JackyWWX, 2023-02-18:
> Fix has been pushed with today's update.

awesome！THX