christianhaitian/arkos #643: RG353M: Reduced sleep mode battery life.

svi-n, 2023-02-17:
### Describe the Issue (If applicable)
In the latest ArkOS update (020923), the device will run the battery out if left in sleep mode overnight. Prior to this release, I've never had the battery die in sleep mode.

I did a quick test, in about one hour of sleep time on the main menu the device used 7% of its battery.

### How can the issue be reproduced? (If applicable)
Charge the device, boot it, and put it into sleep mode (on the main emulation station screen, not in a game).

### What device are you using?
Anbernic RG353M

### What version of ArkOS are you on?
Version (020923)

Thanks for taking the time to look at this!

svi-n, 2023-02-17:
Some more info:
4hrs of sleep time used about ~20% of battery charge.
(wifi/Bluetooth are off)

christianhaitian, 2023-02-17:
Yep this is expected behavior. The default sleep state for ArkOS for 353 devices is now light sleep mode as some units crash when coming out of deep sleep.  You can test if your unit is compatible with deep sleep by enabling deep sleep mode in the options/advanced section.  If you can put the device to sleep and wake it up consistently without experiencing a system crash, then your unit is compatible with deep sleep which has much better battery life.  

Of course, I still recommend turning off the unit completely when not using it for extended periods of time for beet battery life.

closed 2023-02-17 by christianhaitian