christianhaitian/arkos #644: Last update 17/02 for OGA 1.1 broken my control configuration

locodaro, 2023-02-18:
Hi @christianhaitian first of all I want to thank you for your hard work, you have brought my humble oga 1.1 back to life with all your improvements. But this time when the console was restarted after updating to the latest version to date (02/17/2023). When I booted, a message appeared as a pop up, it said that a new control had been detected and that its configuration was mandatory. The problem is that the left stick was misconfigured and I wanted to know how to reappear the controller configuration to fix it.

christianhaitian, 2023-02-18:
Interesting. I think your issue started from 02092023 update.  Something changed with the sdl 2.26.2 update that changed the joystick’s name as emulationstation sees it.  This is the line here < https://github.com/christianhaitian/arkos/blob/cbdf0d55ac53d161e957015bfe525605f74dd7c6/Update-RGB10.sh#L3629>. 

However, that works for the oga 1.1 clone and not the original oga 1.1 it seems.  Waita few mins and I will provide a solution that should fix this for you without needing to do to much.

christianhaitian, 2023-02-18:
Here's an update for Emulationstation that should fix the controls for OGA 1.1 units.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and confirm functionality.

Let me know how it works out for you.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10774370/arkosbackup.tar.gz)

locodaro, 2023-02-18:
Oh my friend **thank you very much** it worked like a charm, thanks again for your quick response, I hope you never stop growing in this and in all the projects that you propose in life. I am very grateful to have your system totally funcional in my oga again. Arkos is excellent, as is the developer who created it. Thanks again mate. Outstanding!

christianhaitian, 2023-02-18:
Awesome.  Thanks for confirming functionality.  Thank you for the kinds words and enjoy!

closed 2023-02-18 by christianhaitian