christianhaitian/arkos #645: Request: Add "Ignore Leading Articles When Sorting" toggle option in UI Settings

Dantasstic, 2023-02-19:
This is a really nice feature in AmberELEC that would be lovely to have on ArkOS as well.

"Legend of Zelda, The" just doesn't have the same ring to it :)

christianhaitian, 2023-02-19:
I'm not quite sure I fully understand how this feature is supposed to work but I just merged it from Batocera's Emulationstation which is what AmberElec uses.  Follow the directions here to load and test this and report back if it works as expected.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and confirm functionality.

again, let me know how it works out.

Thanks,
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10777413/arkosbackup.tar.gz)

christianhaitian, 2023-02-20:
Oh.  I should’ve asked what unit you were using.  Are you familiar with ssh?

Dantasstic, 2023-02-20:
As for how this feature is intended to work, here are some examples.

This is my RG351P running AmberELEC with the "Ignore Leading Articles When Sorting" UI setting turned on. As you can see, this allows games that start with a leading article like the word "The" to show the word at the beginning of the title as intended but still be properly sorted alphabetically by the next word instead. So in this case, "The Legend of Zelda", "The Lion King", etc can be listed properly in the L's, instead of in the T's.

[image]

And here is my RG351V running ArkOS, which currently needs to have the leading article changed to be at the end of the display name to sort properly, so "The Legend of Zelda" and "The Lion King" must become "Legend of Zelda, The" and "Lion King, The".

[image]

Both have the same ROM files on them, just AmberELEC is ignoring the word "The" for sorting purposes.

christianhaitian, 2023-02-20:
For the rk2020 running ArkOS only!

While powering on your device, press and hold the B button down until you boot into the Boot and Recovery Tools menu.  Make sure you have a compatible wireless adapter connected and press the A button on Enable Remote Services.  Once you've done that, you'll need to copy the `emulationstation.header` and `emulationstation` file from the attached zip file to your device using winscp.  Just copy them to the `/home/ark` location.  Once you do that, ssh into the device using a program like [Putty](https://the.earth.li/~sgtatham/putty/latest/w64/putty.exe).  Once you're in the device, copy and paste the following command into the ssh session then press enter.

```
sudo cp /home/ark/emulationstation* /usr/bin/emulationstation/. && sudo chmod 777 /usr/bin/emulationstation/emulationstation* && sudo reboot
```
Once complete, your device should reboot and emulationstation should start up.  Then you can test and confirm the Ignore Leading Articles When Sorting option in the start menu, UI Settings works as expected.  Let me know you results either way.

Thanks again

[emulationstations.zip](https://github.com/christianhaitian/arkos/files/10787744/emulationstations.zip)

christianhaitian, 2023-02-20:
If all works well on the RK2020, you can use the following backup package to update the emulationstation on your ArkOS install for your 351v following the instructions from https://github.com/christianhaitian/arkos/issues/645#issuecomment-1436037166

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10787758/arkosbackup.tar.gz)

Dantasstic, 2023-02-21:
>Once complete, your device should reboot and emulationstation should start up. Then you can test and confirm the Ignore >Leading Articles When Sorting option in the start menu, UI Settings works as expected.

Indeed it does! Thank you for the guide on how to restore the RK2020. I will test it on my RG351V in a little bit and post those results as well.

[image]

Dantasstic, 2023-02-21:
> If all works well on the RK2020, you can use the following backup package to update the emulationstation on your ArkOS install for your 351v following the instructions from [#645 (comment)](https://github.com/christianhaitian/arkos/issues/645#issuecomment-1436037166)
> 
> [arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10787758/arkosbackup.tar.gz)

Yes, it works well! Awesome, thank you for your help! 
Is this option something you would consider adding to the builds going forward?

[image]

christianhaitian, 2023-02-21:
yes.  I'm going to merge them into fcamod since you've confirmed it works as expected and credit you for the request and testing.  Thanks again.

christianhaitian, 2023-02-26:
Officially added with today's update.  Thanks again for suggesting and testing.

closed 2023-02-26 by christianhaitian

Dantasstic, 2023-02-26:
> Officially added with today's update. Thanks again for suggesting and testing.

Thank _you_ for all your hard work and dedication to this project!