christianhaitian/arkos #646: Bluetooth audio stops working after some time in kodi

fpiesik, 2023-02-20:
After a random while of listening (about 1-15 minutes) music in kodi the audio suddenly stops. The connection is still there and the playback timer is keeping on. I can make it work again when I put the device in sleep mode or reconnect by switch the headphones off and on. Sometime it happens the moment I interface with the unit, like changing the volume. Sometimes it habbens just without I do anything but listen. I tried it with two different bluetooth devices. I did experience similar behaviour with playing ps1 emulator.

Reproduce it by  connecting your bluetooth device an start to listen music in kodi. After some time the audio stops (but the playback goes on).

I'm using the rg353m with Arkos version 2.0 (02172023)

christianhaitian, 2023-02-20:
I'll see if I can replicate this issue on my end.

christianhaitian, 2023-03-20:
Hi,

I could not replicate a disconnect issue during my testing using a Philips sb400 or Anker Soundcore 2 true wireless esrbuds using kodi or during gameplay with fbneo.  I do have a new bluetooth audio backend update I will be releasing soon so hopefully it helps with your issue.

closed 2023-03-20 by christianhaitian

tvollbps, 2024-09-16:
I have been having the same issue with my JBL Endurance Race TWS earbuds. I've been using a bluetooth plugin on kodi in order to reconnect the earbuds and correct the audio issue. Using a RGM353M with the most recent version of ArkOS.