christianhaitian/arkos #647: RG353VS - Cannot boot from microSD containing ArkOS.

7st-wa3, 2023-02-21:
Hello.

I'm using RG353VS and
Same symptoms as #620 and I was able to use ArkOS in the same way.

After a while, noise runs on the screen during gameplay and freezes,
I tried resetting and power cycling, but finally it stopped working.

So, format the microSD card again and use Apple Pi Baker
I started over by restoring from ArkOS_RG353V_v2.0_01152023.img.

However, even if I set the completed SD and press and hold the power button, it did not start.

What is this situation and what could be the cause?

By the way, if you have an SD with the official firmware that came with the purchase, it will boot without problems.

The microSD I'm using is SanDisk 32GB UHS-I Class10.
I use Apple Pi Baker on my M1 iMac.

If you have any advice, I would appreciate it.

christianhaitian, 2023-02-21:
Have you fully charged your device?  If so, perhaps see if the same issue happens with UnofficialOS

7st-wa3, 2023-02-21:
Thank you for your prompt reply.

I will fully charge it when I get home and try again and report back.

By the way, does the initial install of ArkOS require the device's battery to be at or near 100% to boot?

7st-wa3, 2023-02-21:
good evening.

I charged the device and when it reached 100%, I tried to start it again.

But it still wouldn't boot.
Even if I press and hold the power button for a long time, even the LED does not light up and the power does not turn on.

(The official firmware boots just fine in the same 100% state.)

RetroGFX, 2023-02-21:
Sounds to me like u may have corrupted the sd card or maybe it's possible that it's a bad flash..I would try different SD cards etc.. And be aware also that Mac's can do weird things work these SD cards add a bunch of unwanted files etc..I would use a Windows PC if possible to flash the sd card.. Use a good brand SD card and also a good program also like win32diskimager to flash it.... If stocks working there's no reason why ark won't work still so something's going on.. The v2 screen stuff has been a pain for all us devs to deal with..I recently just got out first dev images out for uOS for the v2.. like Christan said u could try flashing a different os also.. If that doesn't work you'll know it's a bad flash, card etc

7st-wa3, 2023-02-21:
thank you.

I tried with a different SD card, but the result was the same.

I will try uOS from now on.

7st-wa3, 2023-02-21:
I also tried uOS and the result was the same.

Even after changing the SD card, it still won't boot.

Any firmware other than the official firmware attached to the main unit does not light up the LED and does not start at all.

By the way, the SD that came with the main unit containing the official firmware was once broken, and as instructed by ANBERNIC support, we are rewriting the latest official firmware.

At that time, I used the same M1-iMac and Apple-Pi Baker.

What should I do...

RetroGFX, 2023-02-21:
Do u have a V1 unit or a v2.. the v2 units won't work with Arkos without a few extra steps.. As described in issue 620.. .. So you need to follow the steps to fix it.. But u have a vs so idk if that applies to you.. As for uOS if u have a V1 screen you use the 353p image.. If it's v2 and you don't wanna use the dev image you need to follow the steps in the wiki to get it going.. But I think there's something somewhere you're missing.. Have u tried the steps Christian laid out in issue 620.. u need to test both ways with this unit.. Test it first without switching the dtb then if that doesn't work follow the steps and replace the dtb.. One of those should work for this unit

7st-wa3, 2023-02-21:
As I wrote at the beginning of this thread, I followed the steps in #620 .

So it's been successful once, so I can say my device is v2.

After that, it has become this symptom immediately after it freezes once.

The previous uOS also
I followed the steps described [here](https://github.com/RetroGFX/UnofficialOS/wiki/RG353V-v2-Install-Instructions).

christianhaitian, 2023-02-22:
Try using the stock OS for awhile and make sure it's stable there.  If so, you just may need to keep using that.  If not, look to return the unit.  Good luck.

closed 2023-02-22 by christianhaitian