christianhaitian/arkos #648: ES theme display 2 problems ,

JackyWWX, 2023-02-22:
Recently, I modified an ES theme and found that the image display in svg format would be very blurred. I have tried to change other themes, the same situation .

[image]
compared with JelOS : left-arkos , right-jelos

besides，in gamelist page , If the game description is in Chinese, only one line can be displayed, and it is incomplete.

[image]
compared with JelOS : left-arkos , right-jelos

I love ArkOS，I hope to be able to fix these display problems , just a small suggestion.

christianhaitian, 2023-02-22:
Did you verify that the theme configuration is in 4:3 mode in the start menu?

JackyWWX, 2023-02-23:
> 

This theme was originally made for use by RETROPIE. It has no options and is adaptive to screen ratio. When used on jelos, the image display is normal and not blurred

JackyWWX, 2023-02-23:
> Did you verify that the theme configuration is in 4:3 mode in the start menu?

This theme was originally made for use by RETROPIE. It has no options and is adaptive to screen ratio. When used on jelos, the image display is normal and not blurred

christianhaitian, 2023-02-23:
Then unfortunately, I can't help with this one as the themes used with ArkOS are tuned for the fcamod of emulationstation which is different.  It could be that theme is not compatible with this emulationstation.  Take care.

christianhaitian, 2023-02-23:
maybe the theme setup from [here](https://github.com/Jetup13/es-theme-epicnoir) may give some clues on how you can alter the theme to work but I don't know.

closed 2023-02-23 by christianhaitian