christianhaitian/arkos #649: ArkOS creating unreadable EASYROMS partitions

ylixir, 2023-02-26:
2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

Yes

### Describe the Issue (If applicable)
The EASYROMS partition is unreadable by both Windows 11 and macOS 13.2.1 on both rg353m and rg353v when installed using ArkOS 01152023

### How can the issue be reproduced? (If applicable)
1. Write the ArkOS image to the SD card
    - at this point the EASYROMS folder is available.
2. Insert the SD card into the RG353 and wait for it to finish resizing the partitions
3. Shut down the device and remove the SD card
4. Put the sd card into a Windows or a Mac, it will not have any readable partitions

### What device are you using?
This happens with both RG353V and RG353M

### What version of ArkOS are you on?
01152023

### Anything else you'd like to include that may help to review this issue or feature request?

This is _not_ an unassigned mount point issue that is hammered on throughout the docs. Although from it's prominence in the FAQs, maybe this issue is being confused with that.

The card is using a GPT partition table.

The last partition which _should_ be EASYROMS has no label and is not recognized as a known partition type in either windows 11 or macOS when doing `sudo partx -s /dev/mmcblk1` but the device itself does list it as exfat

When right clicking on the partition in windows, delete is the only option. Interestingly, it's not even listing in the details list.

[image]
[image]

Disk Utility in macOS lists it as "unknown".
[image]

And how the device itself sees things
[image]

christianhaitian, 2023-02-26:
What program are you using to image your sd card?  I can't replicate the issue in Windows 10.

Below is a pic of my OS sd card partitions.
[image]

Even shows the label of EASYROMS for the partition in My Computer
[image]

I don't have a Windows 11 or Mac OS based device so my ability to replicate the issue is limited at this time.

christianhaitian, 2023-02-26:
Minitool partition Wizard in Windows 10

[image]

izrl, 2023-02-26:
Hello! 

I created an account because I've also been encountering this issue. 

I'm using a Windows 11 desktop with all the latest updates installed. I've run into the issue exactly as described. I'm going to copy and paste some of the same things I posted to the RG353V subreddit if that's alright.

> I'm installing ArkOS on a brand new, recently flashed SD card. I followed the guide, and also watched the video, and proceeded with the instructions to the letter. ArkOS is successfully flashed onto the SD card, and the card is able to boot into the operating system in the handheld successfully, I'm able to connect to wifi, etc. Basically pretty much past step 4 of the guide.

Then, when I re-insert the SD card back into my Windows 11 desktop, I am prompted by Windows that the card needs to be formatted. I cancel the prompt, and then proceed to This PC in order to find the EASYROMS folder, however, nothing populates there. I can see that there's something mounted, with an assigned drive letter, but it won't let me proceed without formatting. Which I don't want to do.

When I open disk management, I am able to see the SD card in there, with all the appropriate partitions, the root, boot, easyroms, etc, partitions. However, from disk manager, I am unable to assign a letter to the easyroms partition because its greyed out. Windows will allow me to reassign the drive letter for the ext4 partition, but my understanding is that Windows is unable to read that anyways.

As far as I've observed, I've had no issues with the installation part itself, and I haven't had anything to indicate that I have a V2 screen or that its causing the issue because the symptoms aren't similar.

I've tried the following troubleshooting steps:

    Went through the entire procedure on a different known-good SD card. Same issue persists. No issues following the 2 SD card method, got that working just fine.

    Attempted to use the command prompt to identify and assign a drive letter to the easyroms partition. No dice.

    I was able to use the MiniTool Partition Wizard to assign drive letters to the partition, which allowed me to access the desired partitions. I was able to add bios files, and roms and tested the handheld to be fully functional. However, this is not a permanent solution and will stop working once the SD card is unmounted. So I need to re-do this procedure every time I want to add more games, or make any changes to the easyroms partition.

    I tried following Retro Gaming Corps' video on this issue. I used DiskGenius to prevent Windows from prompting me with the "format this card to view" message, however, the easyroms partition is still unreadable.

    I also checked to see if its my reader creating a problem. I'm using an Anker 575 docking station, and have tried different data cables to make sure that's not an issue. I've also had no issues with other regular UHS-II SD cards for normal functionality. I think its just narrowed down to being an issue under these specific circumstances.

So, I'm here now to ask if I'm missing something terribly obvious or if there's any other advice for me I'd appreciate it.

FWIW, I can make do with having 2 SD cards, or just going through the trouble of constantly using method 3, but I was wondering if there's something I haven't tried yet.

I don't have access to another Windows 11 machine to see if its my OS having some issue or conflict.

I ran into the same issue with the sd card not being readable on MacOS.

christianhaitian, 2023-02-26:
Unfortunately, until I can get access to a Windows 11 install to test and possibly resolve, the only suggestion I have is to use either a 2 sd card setup or copy your games via network to your easyroms partition on sd1.  You can use, samba, winscp, or ArkOS Browser to do this.  See the FAQ for your specific devices in the wiki for more information on using ArkOS Browser.

christianhaitian, 2023-02-26:
I'm creating a new image for the 353 devices.  I made some changes to how the partitions are defined and hope it will help.  I'll post a link to the image here for testing.  Should be available in a couple of hours or so.  Stay tuned.

christianhaitian, 2023-02-26:
Here's a link to the updated 353V/VS image.  Please download and try this and report if anything changes with the unreadable EASYROMS partition.  Thanks,

https://drive.google.com/file/d/1pt7kK2wuN2_cDPwqLy9_3EjfW4WUcrhW/view?usp=share_link

izrl, 2023-02-26:
Thank you! I'll put this on a fresh SD card within the hour and report back with any results or findings.

christianhaitian, 2023-02-28:
Great.  Thanks for confirming👍🏿

xtianstaana, 2023-03-06:
@christianhaitian this seems to be a problem with RG351V as well. The partition comes out as unreadable. Downloaded the (ArkOS_RG351V_v2.0_01152023) version in MEGA.

christianhaitian, 2023-03-06:
I’m working on creating new images with the fix.  Stay tuned

arukaen, 2023-03-06:
> Here's a link to the updated 353V/VS image. Please download and try this and report if anything changes with the unreadable EASYROMS partition. Thanks,
> 
> https://drive.google.com/file/d/1pt7kK2wuN2_cDPwqLy9_3EjfW4WUcrhW/view?usp=share_link

[MEGA](https://mega.nz/file/xfQVxDgT#2v5kRBpFhtBDyKM6AwUtrsFDLzpfHoHJoUfBVYkEEbs)

I want to say that I used this build on my new 353V (purchased 3/4/2023) and it worked great on Windows 11.

CaptainClassic, 2023-03-07:
I can confirm this same issue is affecting the current Gameforce Chi build. I have tried several image writers with the following process:

- Write current ArkOS Chi image to 256gb MicroSD card on Windows 11
- After write, Windows shows BOOT and EASYROMS drives.
- Insert card into ArkOS, which expands EASYROMS as expected.
- After first boot. turn off Chi and reinsert card into Windows 11 computer
- BOOT drive appears but no EASYROMS drive
- Disk Management appears as such
[image]

I've also tried the "Fix exfat partition" option, both before and after updating ArkOS, to no avail.

christianhaitian, 2023-03-07:
yea all the images will have to be updated.  It's in the plans.  Just busy with work at the moment.

CaptainClassic, 2023-03-07:
> yea all the images will have to be updated. It's in the plans. Just busy with work at the moment.

No worries! I was able to use MiniTool Partition Wizard to force-mount the partition. Just wanted to confirm that it wasn't just affecting the RG353.

Thanks for all your hard work on ArkOS, and make sure to take care of yourself as well!

christianhaitian, 2023-03-18:
Updated RG353V and RG353M images with support for reading the BOOT and EASYROMS partitions from Windows 11 and newer Mac OS X are now available from the wiki.

closed 2023-03-18 by christianhaitian

jhanuniv, 2023-06-08:
Hello. The same issue mentioned above also occurs in ArkOS_RG351V_v2.0_03112023. 

Can you provide a new image for RG351V as well?

christianhaitian, 2023-06-09:
did you insert the sd card into the 351v and run through an initial boot first?

jhanuniv, 2023-06-09:
Yes, it exactly matches the symptoms mentioned by ylixir above. The only difference is that my device is 351V.

danftanx, 2023-06-28:
Hi there!  I also created a GitHub account just so that I can follow this thread.  I have the exact same issue described by ylixir on my RG351V.  I thought I was going crazy because I've gone through the process of formatting/imaging/installing ArkOS_RG351V_v2.0_03112023 at least a dozen times on three different SD cards only to have the same issue with the unreadable EASYROMS partition.  Thank goodness I found this thread!  

I will be patiently checking in to see if a new image gets created.  Until then, I'll try the sftp workaround that christianhaitian suggested.  

Also, a big thank you to christianhaitian for his work!  I have no idea how people out there can just go and create OSes like this, but I'm very impressed and grateful!

JayBee-git, 2023-10-06:
Hi there, I am having the exact same issue on the CHI with the latest image, I booted it up so that is resize it all, tried the fix exfat option, tried several PC, sd cards and sd card readers, nothing works, the partition is just not recognized in windows 11.
Is there anything that can be done to fix the issue?