christianhaitian/arkos #651: Portmaster not working correctly on RG353M with 2 SD cards

julian-hoch, 2023-02-27:
Some games work with the ootb settings, e.g. 2048 can be installed and run on /roms2/ports
Other games however seem to try to install to /roms/ports and do not show up in the menu.
I tried to move the ports to /roms/ports and change the directory settings in es_systems.cfg, but not only does that not work either, but then 2048 and other games also will not start anymore.
Specifically I tried to install Stardew Valley and Spelunky, neither of these seem to be working in either /roms/ or /roms2/.

julian-hoch, 2023-02-27:
I have been trying this all morning with various methods, but let me repeat this again for you step by step.
- I just booted up the device (seems it ran out of battery)
- I enable remote services and ssh into the device
- I verify /etc/emulationstation/es_system.cfg and check what the path of "Ports" is - it is currently set to /roms2/ports/
- I check the content of /roms/ports - it is empty
- I check the content of /roms2/ports/ - it contains a few (working) games as well as the PortMaster directory (updated today from github)
- I go to my handheld device and open portmaster via the emulationstation menu (ports => Portmaster (folder) => Portmaster)
- in the portmaster user interface, I go to "All Available Ports" => Spelunky and confirm with Yes
- I can see the system is downloading the package and extracting to /rom/ports
- I check via ssh that indeed /roms/ports now contains a new directory and shell script, while /roms2/ports is unchanged
- i restart emulationstation (start -> quit -> restart emulationstation)
- i go to ports. No new entry is visible in the list.

Anything else you want me to check? I can also walk through what happens if I try to install to /roms/.

christianhaitian, 2023-02-27:
Also, why are you going to ports to open portmaster?  Portmaster is available in the option/tools section

RetroGFX, 2023-02-27:
Also, some games require additional game files to run.. Not all games will work if u just pull them in from portmaster, they require additional files to be installed as described in the wiki.. Hope u get it going.. Good luck.. , good morning Christan

julian-hoch, 2023-02-27:
TLDR: got it to run, thanks!

> You don't state if you used the switch to sd2 script in order to setup the inital sd2 support in ArkOS
I don't remember exactly, since its been a couple of months since I did the initial setup, but in the "options / advanced " menu i see the entry "switch to main SD for Roms", so i assume it is set correctly?

> Also, why are you going to ports to open portmaster? Portmaster is available in the option/tools section.
Interesting, I was not aware of this. I don't remember which guide I followed in the initial setup, but clearly it was incomplete or i missed something there.

> Did you manually install portmaster to your roms/ports folder? If so, that shouldn't have been needed but if you do, it should go into the tools folder.
I did not initially, I think that's how it came? Let me try to fix that.

- I moved the Portmaster folder out of /roms2/ports and into /roms2/tools (overwriting the existing folder, which I guess was older)
- I then wiped /roms/ports again
- I restarted emulation station (ports now does not have the portmaster folder anymore)
- I started portmaster from options/tools. It said there is an update, so I allowed it to update itself. I then restarted it
- I loaded spelunky - this time it loads to /roms2!
- restarted emulationstation and try to start the game. Success!

Now to stardew valley.
- installed it from portmaster
- copy over steamapps/common/Stardew Valley/* to /roms2/ports/stardewvalley/gamedata/ (via sftp)
- restart emulationstation
- checking the contents of the ports menu. There is a new entry, buts all in japanese characters (I think?). It's weird since the shell script looks ok to me. But the game loads up! I guess I can live with this :D

[image]

Anyways, thanks again for your help.

julian-hoch, 2023-02-27:
I figured the last mystery out as well - seems this was just incorrect metadata that probably a previous scrape attempt left behind? after scraping the game again I now have the correct name in the list.

christianhaitian, 2023-02-27:
Awesome.  In the future, the instructions for loading PortMaster for various supported CFWs are located here 
https://github.com/christianhaitian/PortMaster#install-info

Take care and enjoy!

closed 2023-02-27 by christianhaitian