christianhaitian/arkos #652: WiFi seting do not work

lzmvds, 2023-02-27:
The WiFi setting in the emulationstation menu can not scan new network, unless use setting WiFi network in the retroarch seting menu

christianhaitian, 2023-02-27:
Not enough information.  What unit are you using?  ArkOS is available for multiple units.

lzmvds, 2023-02-27:
I am using 351V

christianhaitian, 2023-02-27:
It works fine on that unit with the latest image available for download.  Did you verify your wifi was available?

christianhaitian, 2023-02-27:
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#q-how-do-i-configure-my-wifi-connection-in-arkos

lzmvds, 2023-02-27:
I know the control method, and I'm sure the network is OK. This problem occurred after an update package was updated one day. It was normal before.

christianhaitian, 2023-02-27:
What version of ArkOS are you on?

lzmvds, 2023-02-27:
V2.0(02172023）

christianhaitian, 2023-02-27:
I think I see the issue.  I'll have a resolve for you to try shortly.

christianhaitian, 2023-02-27:
Follow the directions here to load the correct wifi script for the 351v.

For the 351v Only!

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, Launce wifi and verify it functions correctly again

Provide an update with your results as soon as possible.

Thanks,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10841824/arkosbackup.tar.gz)

lzmvds, 2023-02-27:
ok, WiFi scanning is back to normal. Thank you.

lzmvds, 2023-02-27:
Can you also have a look at the script for switching SD card? I switch to TF1 when I am in TF2, I cannot return to TF2, it said that TF2 card was not detected

christianhaitian, 2023-02-27:
When you switch between the sd cards, just reboot afterwards.  Thanks for confirming this fixed your wifi issue.

closed 2023-02-27 by christianhaitian