christianhaitian/arkos #658: Jamma USB pusher interface for two players.

IoannesC, 2023-03-06:
Hello. I was trying to make an arcade machine with an Anbernic RG353V Console. It's very convenient, a portable console and...an arcade machine in the same device (Connecting the screen through HDMi and arcade buttons and joysticks through a Jamma USB connector).

Arkos detect the jamma USB  interface connector (3H Dual Arcade). But only for one player. The jamma interface have two players. The jamma comb connector have two sides (And the USB board too) and in each side have the inputs for every player. Is like a one controller, but for two persons. 

If I try to push the second player buttons, arkos doesn't detect it. For example, in windows I have for every player 12 buttons and a joystick in four directions. But here only for the player one. 

My knowledge is very low in computing. I'm sorry for this and my English is too broken but...

It's possible to make something to use this USB pusher in Arkos? (I tried too in the official firmware and only detect too one player). 

Thanks in advance and thanks for an incredible job in the best anbernic rg353 firmware.

christianhaitian, 2023-03-06:
Try setting up the second player control in the emulationstation menu controller configuration in the start menu.  See if it’s at least recognized there.

christianhaitian, 2023-03-06:
Hmmm....I'm afraid I don't know how to help on this one then.  Sounds like a cool potential use case though.

christianhaitian, 2023-03-06:
Does the Android side recognize player 2?

IoannesC, 2023-03-06:
Let me show you to clarify me...

[image]

This is the USB board. 

[image]

This is an Android input test. The same like in Arkos (Recognize the second joystick like the first, and any second player button works). Maybe I will need to use individual USB boards for each player; a shame. 

[image]
[image]

RetroGFX, 2023-03-06:
I can confirm that emulation station usually requires separate USB boards for the controls..I built my arcade over the summer and had this same issue.. It's very finicky with the inputs for some reason.. Ebay has some good USB control boards for a very decent price..I will also say that my good board was unrecognized and expensive, and the cheap boards I bought to test were both recognized and work great lol.. Go figure..

IoannesC, 2023-03-06:
> I can confirm that emulation station usually requires separate USB boards for the controls..I honestly think it has to do with the control boards itself.. I built my arcade over the summer and had this same issue.. It's very finicky with the inputs for some reason.. Ebay has some good USB control boards for a very decent price..I will also say that my good board was unrecognized and expensive, and the cheap boards I bought to test were both recognized and work great lol.. Go figure..

Yep. Maybe is a hardware issue. I will try with zero delay USB cheap boards :(

christianhaitian, 2023-03-07:
Sorry we couldn't help any further on this.  Good luck!

closed 2023-03-07 by christianhaitian