christianhaitian/arkos #660: [Feature request/question] Any plans to use an up-to-date LTS Ubuntu distribution base instead of EOAN?

eqikkwkp25-cyber, 2023-03-10:
Hello, 

first of all many thanks to the developers of ARKOS. 

Performancewise ARKOS is outstanding to other distributions and apparently EOAN is not an issue here:-) I know that ARKOS is primarily not a Kodi instance but i use it for that purpose too and run into issues with widewine for arm 64 architecture which seems to start working for other LTS distributions recently.  

Are there any plans to do a re-base? 

Thanks.

christianhaitian, 2023-03-10:
Hi,

Not anytime soon.  However, I can look at the widevine issue if there's source code for it.  Maybe the latest code base can be compiled against EOAN to resolve the issues.

Thanks,

closed 2023-03-10 by christianhaitian

eqikkwkp25-cyber, 2023-03-11:
Hello, 

i will open an issue for the widevine problem with some input / links to sources. 

Thanks.