christianhaitian/arkos #662: Issue: RG351v white screen

evicous, 2023-03-12:
### Describe the Issue (If applicable)
Got a new RG351V, looks like a new display panel may be included so there may have been a hardware revision. Looks identical to the previous display issue seen with the RG353V.

### How can the issue be reproduced? (If applicable)
Any new install causes a white screen with a vertical black bar on the left.

### What device are you using?
RG351v

### What version of ArkOS are you on?
the 2.0 package from 01/15

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-03-12:
Does this happen using AmberElec or uOS as well?

evicous, 2023-03-12:
I tried amberelec and lakka. Both showed the same behavior. I can try uOS right now and get back to you.

evicous, 2023-03-12:
Same behavior on uOS.

christianhaitian, 2023-03-12:
Yes they would.  Please provide them along with the boot.ini file.

evicous, 2023-03-12:
[rg351v.zip](https://github.com/christianhaitian/arkos/files/10951057/rg351v.zip)

christianhaitian, 2023-03-12:
Great.  Thank you.

christianhaitian, 2023-03-12:
Try copying the `rk3326-anbernic-st7703.dtb` from the stock OS to the ArkOS installation in the BOOT partition renaming it to `rk3326-rg351v-linux.dtb`.  Make sure to rename the existing `rk3326-rg351v-linux.dtb` file on the ArkOS installation in the BOOT partition to something else first.  Then try to boot ArkOS again and report what happens.

evicous, 2023-03-12:
Looks like that worked! It's went through what I'm assuming is a normal installation process. Booting runs the rumble motor for about 15 seconds. Got to emulationstation, motor stopped, and the display looks correct.

Maybe that rk3326-anbernic-st7703.dtb I uploaded should be packaged in as a separate dtb.v2 file the same as the rg353v.

Thanks for your quick help! Anything else I can do?

christianhaitian, 2023-03-12:
Good and bad problem is that it's too many steps.  Going to see if at least this one can be accomodated without needing to swap that file.  Going to run some tests on my end to see if this can be easily accomodated.  Thanks for your help on this.

evicous, 2023-03-12:
Comment back if you'd like me to test anything in the future! Thanks again

RetroGFX, 2023-03-13:
I really can't believe there's another one of these to deal with.. what a pain .I grabbed that zip for uOS to work on it . I'll get to it soon

leebo, 2023-03-21:
I use this for Jelos. It can boot up, but it doesn't have sound.

rezicG, 2023-03-30:
Might be related, after running into the same issue I attempted this fix and got past the white screen. When trying to connect to wifi, however, no buttons work and I need to reset the device.

rezicG, 2023-03-30:
> Might be related, after running into the same issue I attempted this fix and got past the white screen. When trying to connect to wifi, however, no buttons work and I need to reset the device.

Rebooting while having a Nintendo Switch Pro Controller connected via usbc lets me navigate the wifi menu, though that disables control in the filesystem menu, requiring a reboot without the controller connected for that to work. Again, may or may not be related to the new hardware. Thought it would be worth mentioning.

onlineOften, 2023-04-16:
i'm trying currently to run headless on my RG351V via arkOS. i'm experiencing the white screen on boot. where do i go from here? really looking forward to a response!

RetroGFX, 2023-04-16:
> i'm trying currently to run headless on my RG351V via arkOS. i'm experiencing the white screen on boot. where do i go from here? really looking forward to a response!

Idk what u mean headless, but the white screen can be fixed by downloading this [This dtb](https://drive.google.com/file/d/1e3qvlwpJxJX5CX4gMABvQ2A6rj9sRelx/view?usp=drivesdk) and replacing the one on the ArkOS SD card u have now.. It's the dtb with the fix for the screen.. The directions to replace it are above

onlineOften, 2023-04-16:
> Idk what u mean headless, but the white screen can be fixed by downloading this [This dtb](https://drive.google.com/file/d/1e3qvlwpJxJX5CX4gMABvQ2A6rj9sRelx/view?usp=drivesdk) and replacing the one [...]

still no luck unfortunately

onlineOften, 2023-04-16:
> Idk what u mean headless, but the white screen can be fixed by downloading this [This dtb](https://drive.google.com/file/d/1e3qvlwpJxJX5CX4gMABvQ2A6rj9sRelx/view?usp=drivesdk) and replacing the one [...]

i tried a few times & realized i was doing it completely wrong! it is now working, thank you so much!

RetroGFX, 2023-04-16:
Good to hear . Enjoy

christianhaitian, 2023-04-17:
If possible, can someone test the attached file on a rg351v with a v2 screen.  Hopefully it works and brings working volume button control and wifi.

Make sure to rename the existing rk3326-rg351v-linux.dtb file on the ArkOS installation in the BOOT partition to something else first. Then copy the file within the attached zip file to the BOOT partition and rename it to `rk3326-rg351v-linux.dtb` then try to boot ArkOS again and report what happens.

[rk3326-rg351v-linux-v2.zip](https://github.com/christianhaitian/arkos/files/11254144/rk3326-rg351v-linux-v2.zip)

evicous, 2023-04-17:
I can test this in the next few days, I’ll test volume and wifi as well.

evicous, 2023-04-18:
Using the 2.0_03112023 build, looks like the v2 .dtb works on my RG351V. Screen confirmed working correctly. Volume buttons confirmed working correctly.

I don't use the wifi on any of my emulation devices so I might not be the best person to confirm. I was able to add a wifi connection but that doesn't seem to do anything and wifi in the status screen still shows "down". I think I'm doing it wrong.

Ste-Tis, 2023-04-18:
> If possible, can someone test the attached file on a rg351v with a v2 screen. Hopefully it works and brings working volume button control and wifi.
>
> Make sure to rename the existing [...]

I got a replacement unit with the new screen that is currently running version 2.0 01152023-1. The new file seems to work. The startup is quicker and the device doesn't vibrate that long anymore. The volume button works and also the wifi is up again. My router shows the RG351V as connected device and downloading new themes is also working.

Thanks for the effort!

christianhaitian, 2023-04-18:
Excellent.  Thank you for confirming the functionality.  Next step is to integrate this so that manually changing this file will no longer be necessary.

matschaffer, 2023-04-19:
The above worked for me on AmberElec as well, though volume buttons do not on that firmware. Thanks for the updated driver @christianhaitian !

matschaffer, 2023-04-19:
One more update, the rk3326-rg351v-linux.dtb.v2 found in https://drive.google.com/drive/folders/1kHrVOAlIDt1I9Lkjzp-6Ptigcq2E55OT (from [AmberELEC discord](https://discord.com/channels/777665344289898536/829436688089874464/1096117841964585071)) also works, the volume rockers and wifi work as well.

RetroGFX, 2023-04-19:
> One more update, the rk3326-rg351v-linux.dtb.v2 found in https://drive.google.com/drive/folders/1kHrVOAlIDt1I9Lkjzp-6Ptigcq2E55OT (from [AmberELEC discord](https://discord.com/channels/777665344289898536/829436688089874464/1096117841964585071)) also works, the volume rockers and wifi work as well.

That's my dtb from uOS, mine and Christain's are identical so u know

matschaffer, 2023-04-19:
Interesting! In my last build I renamed `rk3326-rg351v-linux.dtb` to `rk3326-rg351v-linux-orig.dtb` - I wonder if my choice of names might have been what caused the volume rocker problem. Last night I used the discord instructions which said to just overwrite the existing `rk3326-rg351v-linux.dtb`. Just happy to have the volume control back! Thanks so much for the support 🧡

RetroGFX, 2023-04-19:
> Interesting! In my last build I renamed `rk3326-rg351v-linux.dtb` to `rk3326-rg351v-linux-orig.dtb` - I wonder if my choice of names might have been what caused the volume rocker problem. Last night [...]

Honestly who knows lol. There's so much that can go wrong and not play well together.. Esp when we start playing with device trees etc..I notice issues esp when the dtb isn't built with the image from the start . Sometimes it needs to be replaced for some reason.. But the same dtb built with the image has no issues.. Strange

closed 2023-09-27 by christianhaitian

ShippSheroo, 2023-11-03:
I ran into the issue where the screen is white/flashing colors with a black bar on the side as well, leading me to this page but, I don’t know what i’m doing. All the things saying to replace stuff in the BOOT partition, I assumed this meant a BOOT folder but I don’t have one, and I apologize for sounding really dumb right now. I don’t see what anyone is talking about, the root folder is simply EASYROMS (D:) after flashing the 351v img of ArkOS to the MicroSD card. Everything in the root folder is just folders for other emulators.

I wasn’t expecting to get in over my head with technical issues. All the research I did before buying this device made everything look as simple as dragging and dropping or using programs to flash but I keep running into problems. I’ve modded all my retro consoles just fine without any issues like this (Wii, Vita, PSP, 3DS)

christianhaitian, 2023-11-03:
> I ran into the issue where the screen is white/flashing colors with a black bar on the side as well, leading me to this page but, I don’t know what i’m doing. The solutions above saying replace some files in the BOOT partition, I assumed this meant a BOOT folder but I don’t have one, and I apologize for sounding really dumb right now. I don’t see what anyone is talking about, the root folder is simply EASYROMS (D:) after flashing the 351v img of ArkOS to the MicroSD card. Everything in the root folder is just folders for other emulators.
> 
> I wasn’t expecting to get in over my head with technical issues. All the research I did before buying this device made everything look as simple as dragging and dropping or using programs to flash but I keep running into problems. I’ve modded all my retro consoles just fine without any issues like this (Wii, Vita, PSP, 3DS)

It usually is as simple as flashing an OS to a sd card but the manufacturer has changed the screen on this device due to supply issues and has not shared how to boot more than 1 screen type in a single image.  So what you have to do for what we call a v2 screen is to replace a dtb file that is on the boot partition of the sd card once you've flashed ArkOS to a MicroSD card.  If you're only seeing one partition, then either Windows has not assigned a drive letter to your fat32 boot partition for the sd card, or you used Balena Etcher to flash the sd card which causes issues.

Once you can see the fat32 partition, it should be as simple as following the directions here: (https://github.com/christianhaitian/arkos/issues/662#issuecomment-1511860062) to replace a single file than boot the device.

ShippSheroo, 2023-11-03:
> > I ran into the issue where the screen is white/flashing colors with a black bar on the side as well, leading me to this page but, I don’t know what i’m doing. The solutions above saying replace some files in the BOOT partition, I assumed this meant a BOOT folder but I don’t have one, and I apologize for sounding really dumb right now. I don’t see what anyone is talking about, the root folder is simply EASYROMS (D:) after flashing the 351v img of ArkOS to the MicroSD card. Everything in the root folder is just folders for other emulators.
> > I wasn’t expecting to get in over my head with technical issues. All the research I did before buying this device made everything look as simple as dragging and dropping or using programs to flash but I keep running into problems. I’ve modded all my retro consoles just fine without any issues like this (Wii, Vita, PSP, 3DS)
> 
> It usually is as simple as flashing an OS to a sd card but the manufacturer has changed the screen on this device due to supply issues and has not shared how to boot more than 1 screen type in a single image. So what you have to do for what we call a v2 screen is to replace a dtb file that is on the boot partition of the sd card once you've flashed ArkOS to a MicroSD card. If you're only seeing one partition, then either Windows has not assigned a drive letter to your fat32 boot partition for the sd card, or you used Balena Etcher to flash the sd card which causes issues.
> 
> Once you can see the fat32 partition, it should be as simple as following the directions here: ([#662 (comment)](https://github.com/christianhaitian/arkos/issues/662#issuecomment-1511860062)) to replace a single file than boot the device.

I used Rufus as the tutorial instructed, as it did mention the problems with Balena Etcher. I’ll try Win32DiskImager that is also linked in the tutorial. Thank you for your fast reply, I hope I can get it figured out

AJJOHNSON17, 2024-01-06:
hi guys, followed this feed to get past the white screen issue but still having the sound rocker issue on Amberelec, was there a fix for this operating system at all? any help is much appreciated.

AJJOHNSON17, 2024-01-08:
Sorry just had a re-read and seen this has already been answered and it has fixed it, thanks for the thread guys, you life savers.

wavetro, 2024-12-01:
Not to re-open this issue but I wanted to add an updated fix for anyone still struggling to fix the white or yellow screen issue on your RG351V, even after swapping out the dtb file in boot with the v2 from this thread:

1. DON'T use the .dtb files from this thread
2. Flash the SD card with the latest ROM from [Anbernic's website](https://win.anbernic.com/download/317.html) instead
2. Go to the "boot" partition and copy the `rk3326-anbernic-st7703.dtb` file off the card onto your computer
3. Flash the SD card with the latest ArkOS from the wiki like usual
4. Rename the file you copied from the original Anbernic ROM to `rk3326-rg351v-linux.dtb` and paste it into the "boot" partition of your ArkOS card, replacing the old one

It should work now

tony8607, 2025-01-29:
opensim hardware osh pb controller 4617/12544 not configured everytime i try to play a game and i need the right  latest prerelease the release image is very old

evnb, 2025-03-08:
I followed these instructions and the device is usable now, but the screen is still flashing with that vignetted off-center white screen.  Does anyone know how to fix? 

[image]

evnb, 2025-03-09:
> I followed these instructions and the device is usable now, but the screen is still flashing with that vignetted off-center white screen. Does anyone know how to fix?
I had renamed the original `rk3326-rg351v-linux.dtb` on the BOOT drive to `rk3326-rg351v-linux.dtb.v0`. It seems that there was some issue with that still being used.

Fix: make sure to change the name *before* the extension. Eg `rk3326-rg351v-linux-v0.dtb.v0`. That got rid of the flickering screen for me!

pbanj, 2025-03-09:
> Not to re-open this issue but I wanted to add an updated fix for anyone still struggling to fix the white or yellow screen issue on your RG351V, even after swapping out the dtb file in boot with the v2 from this thread:
> 
>     1. **DON'T** use the .dtb files from this thread
> 
>     2. Flash the SD card with the latest ROM from [Anbernic's website](https://win.anbernic.com/download/317.html) instead
> 
>     3. Go to the "boot" partition and copy the `rk3326-anbernic-st7703.dtb` file off the card onto your computer
> 
>     4. Flash the SD card with the latest ArkOS from the wiki like usual
> 
>     5. Rename the file you copied from the original Anbernic ROM to `rk3326-rg351v-linux.dtb` and paste it into the "boot" partition of your ArkOS card, replacing the old .dtb file of the same name that comes with ArkOS
> 
> 
> It should work now

or you could post a link to the file and save all that wear and tear on the sd card
https://z01.zip/rk3326-rg351v-linux.dtb

evnb, 2025-03-12:
> > Interesting! In my last build I renamed `rk3326-rg351v-linux.dtb` to `rk3326-rg351v-linux-orig.dtb` - I wonder if my choice of names might have been what caused the volume rocker problem. Last night I used the discord instructions which said to just overwrite the existing `rk3326-rg351v-linux.dtb`. Just happy to have the volume control back! Thanks so much for the support 🧡
> 
> Honestly who knows lol. There's so much that can go wrong and not play well together.. Esp when we start playing with device trees etc..I notice issues esp when the dtb isn't built with the image from the start . Sometimes it needs to be replaced for some reason.. But the same dtb built with the image has no issues.. Strange .. And thanks for the support

I just tested it and using the v2 file is the only way I've been able to get volume rocker to work on arkos on my 351v. I tested with and without an extra rk3326-rg351v-linux-**v0**.dtb.v0 file in the BOOT drive, and they both worked fine. I believe that adding a file named rk3326-rg351v-linux.dtb.v0 would result in the blinking image I showed above.

GautreauLuke, 2025-04-17:
> I just tested it and using the v2 file is the only way I've been able to get volume rocker to work on arkos on my 351v. I tested with and without an extra rk3326-rg351v-linux-**v0**.dtb.v0 file in [...]

Where did you get the v2 file that fixed the volume rocker issue? I'm running AmberElec on mine right now, and it seems to be running fine and without flickering using the dtb file posted above, but I'm still having volume rocker issues.

apetrella90, 2025-05-10:
I can't access the boot partition no matter what I do. I truly have no idea what I am doing wrong.

chronoss09, 2025-07-26:
> I just tested it and using the v2 file is the only way I've been able to get volume rocker to work on arkos on my 351v. I tested with and without an extra rk3326-rg351v-linux-**v0**.dtb.v0 file in [...]

It work but same issue with other about the volume rocker :(

chronoss09, 2025-08-10:
> > > > > Interesting! In my last build I renamed `rk3326-rg351v-linux.dtb` to `rk3326-rg351v-linux-orig.dtb` - I wonder if my choice of names might have been what caused the volume rocker problem. Last night I used the discord instructions which said to just overwrite the existing `rk3326-rg351v-linux.dtb`. Just happy to have the volume control back! Thanks so much for the support 🧡
> > > > 
> > > > 
> > > > Honestly who knows lol. There's so much that can go wrong and not play well together.. Esp when we start playing with device trees etc..I notice issues esp when the dtb isn't built with the image from the start . Sometimes it needs to be replaced for some reason.. But the same dtb built with the image has no issues.. Strange .. And thanks for the support
> > > 
> > > 
> > > I just tested it and using the v2 file is the only way I've been able to get volume rocker to work on arkos on my 351v. I tested with and without an extra rk3326-rg351v-linux-**v0**.dtb.v0 file in the BOOT drive, and they both worked fine. I believe that adding a file named rk3326-rg351v-linux.dtb.v0 would result in the blinking image I showed above.
> > 
> > 
> > Where did you get the v2 file that fixed the volume rocker issue? I'm running AmberElec on mine right now, and it seems to be running fine and without flickering using the dtb file posted above, but I'm still having volume rocker issues.
> 
> IIRC I got it from this thread. (Or maybe the discord?) I just don't remember which comment contains the file that worked for me.

can u share the file please ?