christianhaitian/arkos #664: GZDoom not started after updates

Pash-tet, 2023-03-13:
After updates i have error on gzdoom start. Console output:
`
W_Init: Init WADfiles.
 adding /home/ark/.config/gzdoom/gzdoom.pk3, 667 lumps
 adding /home/ark/.config/gzdoom/game_support.pk3, 2514 lumps
 adding /home/ark/.config/gzdoom/PLUTONIA.WAD, 2988 lumps
 adding /home/ark/.config/gzdoom/game_widescreen_gfx.pk3, 214 lumps
I_Init: Setting up machine state.
V_Init: allocate screen.
S_Init: Setting up sound.
I_InitSound: Initializing OpenAL
  Opened device ALSA Default
  EFX enabled
ST_Init: Init startup screen.
Checking cmd-line parameters...
S_InitData: Load sound definitions.
G_ParseMapInfo: Load map definitions.
Texman.Init: Init texture manager.
ParseTeamInfo: Load team definitions.
LoadActors: Load actor definitions.
script parsing took 1376.25 ms
R_Init: Init Doom refresh subsystem.
DecalLibrary: Load decals.
M_Init: Init menus.
P_Init: Init Playloop state.
ParseSBarInfo: Loading custom status bar definition.
D_CheckNetGame: Checking network game status.
player 1 of 1 (1 nodes)
Using video driver KMSDRM

Only SDL 2.0.6 or later is supported.
`

christianhaitian, 2023-03-13:
Odd.  BD Lite starts up with GzDoom fine but Doom 1 and Doom 2 don't.  I'll investigate and see what's going on.

christianhaitian, 2023-03-13:
Doesn't like the new SDL 2.26.  Testing to see if a newer gzdoom build will resolve this.

christianhaitian, 2023-03-13:
Hi, The latest build of gzdoom seems to work fine on my end.  Can you help test and let me know if it works for you as well?

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, you should be able to launch gzdoom.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10960640/arkosbackup.tar.gz)

Pash-tet, 2023-03-13:
> Hi, The latest build of gzdoom seems to work fine on my end. Can you help test and let me know if it works for you as well?
>
> 1. Take the attached compressed file (arkosbackup.tar.gz) and put it in [...]

I did everything according to your instructions, now it seems that some library is missing. When trying to start output:
```
./gzdoom: error while loading shared libraries: libvpx.so.5: cannot open shared object file: No such file or directory
```

christianhaitian, 2023-03-13:
Oops, I forgot to include this library. Try this one instead.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10960797/arkosbackup.tar.gz)

Pash-tet, 2023-03-13:
> Oops, I forgot to include this library. Try this one instead.
> 
> [arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10960797/arkosbackup.tar.gz)

Now it works like a charm. Thank you. 

I only have one little problem: now then I'm launching any .doom file with IWAD= line, launching wad I selected when I tested  doom.sh, and not which is registered in the .doom file in the roms/doom directory)

christianhaitian, 2023-03-13:
Post the contents of your .doom file.

Pash-tet, 2023-03-13:
> defaultiwad

IWAD=iwads/HERETIC.wad
MOD=mods/brutal_heretic.pk3
MOD=mods/aim_assist.pk3
-- end --

closed 2023-03-17 by christianhaitian