christianhaitian/arkos #666: GameCube / Dolphin not showing up

dvbavel, 2023-03-16:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

After installing dolphin via RetroArch -> Core System Files Downloader and adding an .ciso file to /roms2/gc dolphin / gamecube is not showing up in the frontend after reboot of device 

### How can the issue be reproduced? (If applicable)

Setup 2 sd cards

Sd card 1: ArkOS
Sd card 2: Roms / Bios

Used advanced -> switch to .... (roms2) for roms (don't know how it's exactly named before the switch)

Downloaded Dolphin.zip via retroarch and added an .ciso to /roms2/gc

Rebooted device

Nothing to be found.

### What device are you using?

RG353V 

### What version of ArkOS are you on?
2.0 03112023

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-03-17:
Did you check the ArkOS Emulator and Ports Information for supported extensions?
https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#gamecube-rg353m-rg353v-rg353vs-and-rg503-only

dvbavel, 2023-03-17:
Hi @christianhaitian yes like an 100 times, but I don't see anything about Dolphin installation/ enabling tbh. Could you perhaps throw coffee in my eyes and point me to the specific thing your meaning?

christianhaitian, 2023-03-17:
Emulator is already pre installed.  .ciso is not a supported extension.

dvbavel, 2023-03-17:
Ah like that damn fully oversaw ciso is not there. Fully confused due to https://github.com/libretro/libretro-core-info/blob/master/dolphin_launcher_libretro.info and Dolphin community talking bout .ciso files. 

Many thanks for opening the eyes and pointing it to my face! ArkOS expects the .nkit.iso format then?

I'll try this later - again many thanks for all your work and efforts!

dvbavel, 2023-03-17:
So it indeed was an layer 8 problem running in too an pebkac due too failing on RTFM. Many thanks for your support @christianhaitian closing this one.

closed 2023-03-17 by dvbavel