christianhaitian/arkos #667: Dolphin not taking any input of RG353V keypad

dvbavel, 2023-03-17:
### Describe the Issue (If applicable)
As a follow up of my layer 8 problem resulting in PEBKAC (https://github.com/christianhaitian/arkos/issues/666) it must be the number... I managed to startup GC games but none of them is taking any input, start, YXAB nothing is working. How to reset the Dolphin key config? How to make it work with the RG353V keypad? I'm again convinced I did read all your Wiki pages, Issues, etc.. And I couldn't find it, managing Dolphin in general seems to be an impossible search query.

Any other games work fine and input is just working, just Dolphin seems to be failing.

### How can the issue be reproduced? (If applicable)
Start an GC game and try giving it input

### What device are you using?

RG353V

### What version of ArkOS are you on?
2.0 03112023

### Anything else you'd like to include that may help to review this issue or feature request?

duskgao, 2023-03-17:
Which game did you test it with? Is it possible that you're testing a Wii game? I've found that NGC games that use controllers work, like SRW GC, but those Wii games that use controllers don't, like SRW Neo

dvbavel, 2023-03-17:
I did try with Pikmin 2, Metroid Prime, Mario Kart Double Dash, Paper Mario. Got them from an certain not to name website under catagory GameCube. All those games are to my knowledge all real NGC games.

christianhaitian, 2023-03-17:
Paper Mario was to slow to run but I remember the controls working fine there.  I also have Pikmin 1 working fine on my end.  Did you make any changes to dolphin configurations via linux filesystem?

dvbavel, 2023-03-17:
Yeh I guess most games will run crap, but I can't even skip the splashscreen as none of the buttons are doing anything.

No didn't change anything in the config, I've openend it though but did not change a thing in it.

Can I safely purge the config folder? Would it recreate the files?

christianhaitian, 2023-03-17:
No.  Make sure `/home/ark/.local/share/dolphine-emu/Config/GCPadNew.ini` contains the following:

```
[GCPad1]
Device = evdev/0/retrogame_joypad
Buttons/A = Button 0
Buttons/B = Button 1
Buttons/Start = Button 9
Buttons/X = Button 3
Buttons/Y = Button 2
Buttons/Z = Button 7
Buttons/Hotkey = Button 8
C-Stick/Dead Zone = 25.000000000000000
C-Stick/Down = Axis 3+
C-Stick/Left = Axis 2-
C-Stick/Modifier = Control_L
C-Stick/Modifier/Range = 50.000000000000000
C-Stick/Right = Axis 2+
C-Stick/Up = Axis 3-
D-Pad/Down = Button 14
D-Pad/Left = Button 15
D-Pad/Right = Button 16
D-Pad/Up = Button 13
Main Stick/Dead Zone = 25.000000000000000
Main Stick/Down = Axis 1+
Main Stick/Left = Axis 0-
Main Stick/Modifier = Shift_L
Main Stick/Modifier/Range = 50.000000000000000
Main Stick/Right = Axis 0+
Main Stick/Up = Axis 1-
Triggers/L = Button 4
Triggers/R = Button 5
```

closed 2023-03-18 by christianhaitian