christianhaitian/arkos #668: Bluetooth and other options are disabled once you turn on "Parse gamelists only"

evillurker, 2023-03-17:
If you turn on "Parse gamelists only" in advanced options, many options (including bluetooth) will not show up anymore.
This is necessary for DOS games.

christianhaitian, 2023-03-17:
Parse gamelists only is only useful to just save a few seconds to boot into ArkOS.  This feature usually causes more issues than it helps since most people forget when they turn this on then wonder why new games they add to the system don't show up.  Why do you feel this is this necessary for DOS games?  Many options would disappear with this on since the options section is not parsed in order to prevent those options from showing up in the Last Played collections.

evillurker, 2023-03-17:
If I don't use this for dos games, the files inside the folders themselves will show up on the playlist. There is no other way to make it show properly without using this feature. If there is another way, let me know

christianhaitian, 2023-03-17:
You mean pure.zip files?  If so, rename your .zip dos files to .dosz then hide .zip msdos files.

closed 2023-03-18 by christianhaitian