christianhaitian/arkos #671: RG353V ArkOS file corrupted after extraction

effluvialsplurf, 2023-03-23:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

After download the zip file is corrupted

### How can the issue be reproduced? (If applicable)

downloading the file from either mega or gdrive
### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?
I have tried both download sources and neither supplies a non-corrupted download

christianhaitian, 2023-03-23:
They are not zip files.  They are .xz files and can be opened using 7zip.

effluvialsplurf, 2023-03-23:
I have opened them, they are corrupted immediately upon download

effluvialsplurf, 2023-03-23:
I have 7zip installed and I have tried extracting the img file from the xy file but it is also corrupted

effluvialsplurf, 2023-03-23:
I apologize I forgot to change the title

christianhaitian, 2023-03-23:
I just downloaded them and tested them again and they open just fine using 7zip wtihout issue.  Are you sure it's not an issue with your computer?

christianhaitian, 2023-03-23:
Do you have antivirus software that is flagging the file maybe?

effluvialsplurf, 2023-03-23:
I dont have antivirus aside from default windows defender, the only thing I could think on that end would be my firewall but I dont think that would cause a downloaded file to be corrupted.

effluvialsplurf, 2023-03-23:
could it be that I use microsoft edge?

christianhaitian, 2023-03-23:
I don't know.  Maybe try a different browser.  I use Google Chrome and Firefox myself.  Also, when download is complete, verify that the md5 is the same as what's listed on the wiki for the file.  If they are not, that means something is altering the file during your download.

effluvialsplurf, 2023-03-23:
ok give me a min

effluvialsplurf, 2023-03-23:
its giving me a totally different md5

f77c99c4830744575ffc75e3e7843a71

would it matter how off it is or is it just about matching or not?

effluvialsplurf, 2023-03-23:
but I suppose this answers my question, I suppose its probably my internet, but if you think it could be something else please let me know, if not feel free to close the ticket and I'll figure it out.