christianhaitian/arkos #672: Freezes on resume.

pikamus, 2023-03-23:
### Describe the Issue (If applicable)
It appears to only happen with i'm in the emulation station UI, in a game category and using detailed view.

### How can the issue be reproduced? (If applicable)
I have a feeling it only happens when the game has a video preview, and i put it to sleep on a game.

### What device are you using?
Rg353v

### What version of ArkOS are you on?
Latest as of 23/03/2023

christianhaitian, 2023-03-23:
Known issue with ES and video previews.  However, it shouldn't freeze in ArkOS.  It should stop ES before going to sleep, the nreload ES on resume.  How long ago did you load ArkOS and have you been updating via OTA regularly?

pikamus, 2023-03-23:
I update OTA pretty much every version.
I try to shutdown Arkos fully from time to time to conserve battery.
It does fully freeze my arkos, so I have to usually press the console reset button.

christianhaitian, 2023-03-23:
Hmmm...I can't replicate the issue.  When I set a system to detailed view, no video previews play.  When I leave it on automatic, video previews play and when I tap the power button, it shutdown emulationstation, then sleeps the unit.  When power button is pressed again, ES starts backup.  It's working as expected on my end.  Did you change the sleep mode of the device to deep sleep instead of light sleep?

christianhaitian, 2023-03-23:
in detailed view, within a game category, I can sleep and wake the device multiple times without issue.

pikamus, 2023-03-23:
ahh yea, I did put the device is on Deep Sleep mode.
using theme Arkos-carbon
gamelist view style detailed.
system view  legacy, system background carbon legacy

pikamus, 2023-03-23:
switching to Light sleep mode fixes the issue

christianhaitian, 2023-03-23:
Ok.  Just means your device can’t do deep sleep reliably. Take care.

closed 2023-03-23 by christianhaitian