christianhaitian/arkos #674: [Feature request] Provide Kodi / Xmbc  inputstream.adaptive addon

eqikkwkp25-cyber, 2023-03-24:
Hello, 

unfortunately my attempt to compile 

[https://github.com/xbmc/inputstream.adaptive](https://github.com/xbmc/inputstream.adaptive)

using comment [https://github.com/xbmc/inputstream.adaptive/pull/1166#issuecomment-1474785256](https://github.com/xbmc/inputstream.adaptive/pull/1166#issuecomment-1474785256)

on device RG503 were not successful, i.e. i ran into different kind of library dependency issues. 

Can you provide or include this addon so that we can use widewine for different kind of Kodi addons like Netflix etc.? Thanks in advance.

christianhaitian, 2023-03-24:
Building something now on this.  If it completes successfully, I'll post it here for you to test.

christianhaitian, 2023-03-24:
Hi,

Attached is the addon built to be installable in Kodi 19.5 for ArkOS.  I was able to install the zip without issue using Kodi on my RG353V.  Hopefully it works for the application you need.  Let me know your results when possible.

Thanks,

[inputstream.adaptive.zip](https://github.com/christianhaitian/arkos/files/11066525/inputstream.adaptive.zip)

eqikkwkp25-cyber, 2023-03-25:
Many thanks. If you followed their build instructions [https://github.com/xbmc/inputstream.adaptive/wiki/How-to-build](https://github.com/xbmc/inputstream.adaptive/wiki/How-to-build) and on top the patch for widevine arm64 you can close this issue. 

For me the above addon zip installs on RG503 and is visible within Arkos Kodi / Matrix. Everything else, i.e. manual install of widevine arm64 binary seems a little bit messy resp. magic as it does not seem to work for all users the same way. There is work in progress with regards inputstreamhelper which downloads an image and extracts the widewine lib, see [https://github.com/emilsvennesson/script.module.inputstreamhelper/pull/533](https://github.com/emilsvennesson/script.module.inputstreamhelper/pull/533). If other ARKOS users succeed please drop a note here.

christianhaitian, 2023-03-25:
Seemed like the patch wasn't necessary as it was merged into the code base itself.  If something changes on this, let me know.  I plan to eventually update Kodi to version 20 once it matures a bit more and more plugins work with it.

closed 2023-03-25 by christianhaitian

eqikkwkp25-cyber, 2023-03-25:
> Seemed like the patch wasn't necessary as it was merged into the code base itself. If something changes on this, let me know. I plan to eventually update Kodi to version 20 once it matures a bit more and more plugins work with it.

Yes, for Kodi 20 and Kodi 21 the patch was merged via PR but not for Matrix. Most distributions are still using Matrix afaik, i.e. it makes sense to wait before updating it within ArkOS. 

I guess you are cross compiling ArkOS and binaries. Is there some kind of SDK out or an instruction?

christianhaitian, 2023-03-25:
For the Kodi build, I use this: https://github.com/christianhaitian/kodi-install

christianhaitian, 2023-03-25:
If you're building on the device and not through a chroot or docker, have you setup dev mode on the unit yet?  If not, you'll need to do that.  Be forewarned that it wipes the contents of your roms partition so you should use a separate sd card for that or backup your roms folder content.

https://github.com/christianhaitian/arkos/wiki/Building-packages-and-modules-on-your-device

eqikkwkp25-cyber, 2023-03-25:
> 

That does not work for me "Couldn't install an important package from apt. Are you connected to the internet?" 

```
ark@rg503:~/devel$ ./install_headers.sh 
librga:RGA_GET_VERSION:3.02,3.020000 
ctx=0x55887d1eb0,ctx->rgaFd=3 
Rga built version:version:+2017-09-28 10:12:42 
obj=0x55690500c0, event=12
obj=0x55690500c0, event=11
obj=0x55690500c0, event=0
obj=0x55690500c0, event=3
obj=0x55690500c0, event=6
obj=0x55690500c0, event=7
librga:RGA_GET_VERSION:3.02,3.020000 
ctx=0x558c0b9eb0,ctx->rgaFd=3 
tee: '': No such file or directory
GODEV
Hit:1 http://ppa.launchpad.net/team-xbmc/ppa/ubuntu eoan InRelease
Hit:2 http://old-releases.ubuntu.com/ubuntu eoan InRelease
Hit:3 http://old-releases.ubuntu.com/ubuntu eoan-updates InRelease
Hit:4 http://old-releases.ubuntu.com/ubuntu eoan-security InRelease
Hit:5 http://deb.odroid.in/go2 bionic InRelease  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cloud-guest-utils is already the newest version (0.31-5-gef42f6b5-0ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up bootini (20201213-8) ...
mount: /media/boot: special device /dev/mmcblk0p1 does not exist.
umount: /media/boot: not mounted.
dpkg: error processing package bootini (--configure):
 installed bootini package post-installation script subprocess returned error exit status 32
Errors were encountered while processing:
 bootini
E: Sub-process /usr/bin/dpkg returned an error code (1)
librga:RGA_GET_VERSION:3.02,3.020000 
ctx=0x555ec75eb0,ctx->rgaFd=3 
Rga built version:version:+2017-09-28 10:12:42 
obj=0x555dd36f40, event=12
```

christianhaitian, 2023-03-25:
I just tested this on a fresh install of ArkOS on my RG503.  I created a devel subfolder and followed the instructions and it worked just fine for me.  Did you make sure your internet connection worked?

eqikkwkp25-cyber, 2023-03-26:
> I just tested this on a fresh install of ArkOS on my RG503. I created a devel subfolder and followed the instructions and it worked just fine for me. Did you make sure your internet connection worked?

Internet connection worked. I just re-tested with a fresh install and enabling developer mode works, including compiling some stuff. Thanks.