christianhaitian/arkos #676: Celeste (Full) doesn't work on ArkOS v2.0.03.21.2023 / RG353V

CIPop, 2023-03-25:
### Describe the Issue (If applicable)
I'm trying to run [Celeste](https://retrogamecorps.com/2022/04/30/guide-celeste-on-the-anbernic-rg552/) Full (not pico-8) on ArkOS latest.

### How can the issue be reproduced? (If applicable)

Follow the instructions to install at https://retrogamecorps.com/2022/04/30/guide-celeste-on-the-anbernic-rg552/ 
Game starts, displays "Loading" then quits. (I don;t kn

### What device are you using?
RG353V

### What version of ArkOS are you on?
v2.0.03.21.2023

### Anything else you'd like to include that may help to review this issue or feature request?
This works fine on:
- 353V Stock latest
- 353V Jelos latest

(The port binaries were created on AnberElec/351Elec on an 351V and copied over to 353V.)

Other ports such as Iconoclasts, Half-Life 1 work as expected on ArkOS.

A comment from https://retrogamecorps.com/2022/04/30/guide-celeste-on-the-anbernic-rg552/ is suspecting an old SDL2 lib:

> Note that the “Final” version of ArkOS for the RG351P and RG351M will not support the game due to an outdated SDL2 library. 

There is a log in celeste/savedata:

```
Celeste Error Log
==========================================

Ver 1.4.0.0
3/24/2023 9:55:26 PM
System.IO.EndOfStreamException: Unable to read beyond the end of the stream.
  at System.IO.MemoryStream.InternalReadInt32 () [0x00035] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at (wrapper remoting-invoke-with-check) System.IO.MemoryStream.InternalReadInt32()
  at System.IO.BinaryReader.ReadInt32 () [0x00027] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at Celeste.PlaybackData.Import (System.Byte[] buffer) [0x00093] in <d7b2abb405ec4c97954549f0a02e4a41>:0 
  at Celeste.PlaybackData.Load () [0x00028] in <d7b2abb405ec4c97954549f0a02e4a41>:0 
  at Celeste.Celeste.LoadContent () [0x0001a] in <d7b2abb405ec4c97954549f0a02e4a41>:0 
  at Microsoft.Xna.Framework.Game.Initialize () [0x0007f] in <0f64a64991244c77b1013be94c53aca1>:0 
  at Monocle.Engine.Initialize () [0x00000] in <d7b2abb405ec4c97954549f0a02e4a41>:0 
  at Celeste.Celeste.Initialize () [0x00000] in <d7b2abb405ec4c97954549f0a02e4a41>:0 
  at Microsoft.Xna.Framework.Game.DoInitialize () [0x0011d] in <0f64a64991244c77b1013be94c53aca1>:0 
  at Microsoft.Xna.Framework.Game.Run () [0x00011] in <0f64a64991244c77b1013be94c53aca1>:0 
  at Monocle.Engine.RunWithLogging () [0x00000] in <d7b2abb405ec4c97954549f0a02e4a41>:0 
```

This made me think it might be a path incorrectly set up?

christianhaitian, 2023-03-25:
Celeste runs fine on my end so I know it works.  Are you using the pre-included portmaster files that's available from the Options/Tools section?  If not, please use that and do not load portmaster into the ports folder.  That can cause issues like what you describe in ArkOS.

christianhaitian, 2023-03-28:
Any update on this?

CIPop, 2023-03-29:
I think that's it: I had an AnberElec SD and just copied the Ports folder over.
Afterwards I ran PortMaster from ports, not from options to bring in the ports.

I'll start over with a clean SD card and update here.

closed 2023-04-01 by christianhaitian