christianhaitian/arkos #677: RG353V V2 unable to install after v2 file replacement - Black screen

OldJer, 2023-03-25:
### Describe the Issue (If applicable)
Black screen after v2 file replacement, no logo. System appears hung.

### How can the issue be reproduced? (If applicable)
Image the latest RG353V issue to an SD card (v2.0 3-11-2023) as of time of writing.

Do the OC v2 file replacement in the BOOT partition after getting the garbled screen known issue.

Screen will be black on next boot. Device LED goes red and appears to hang. No logo or any indication that system is advancing through boot process. Have attempted with several name brand SD cards and also preemptively doing the v2 file replacement before initial boot. Same black screen.

### What device are you using?
RG353V v2

### What version of ArkOS are you on?
Attempting to install 3-11-2023

### Anything else you'd like to include that may help to review this issue or feature request?
I’ve attempted using multiple imaging programs (Linux Mint’s built in USB Image Writer and balenaEtcher) and used a couple Sandisk 32gb cards and a Samsung card with no change.

christianhaitian, 2023-03-26:
I tested and verified on my end that renaming the .v2 file to rk3566-OC.dtb works on my rg353v v2 unit.  Have you tried UnofficialOS to see if that boots for you?  If it does not boot and but the stock linux os does, it could be Anbernic may have changed screens again.

RetroGFX, 2023-03-26:
Are u sure your unit is def a v2?? If u didn't get it directly from anbernic it might be a V1 still

christianhaitian, 2023-03-26:
Yes, my unit is a v2 that was sent to me for development by Anbernic.  My screen doesn’t work with the default insge as is without making that file change.

OldJer, 2023-03-26:
> 

UnofficialOS starts to boot after making that v2 file change and editing the conf but never gets beyond “initializing system, please wait.” Stock linux boots and runs fine on both the supplied SD and one of the Sandisk SDs using 01082023.

OldJer, 2023-03-26:
Now I can actually mark this resolved. Grabbed my WIndows laptop and used the imaging tool recommended in the Wiki (USB Image Tool). Made the v2 file replacement and the image fired right up. Looks like the image tool built into Linux Mint may be a little flaky. My Mint PC is running 21.1.

christianhaitian, 2023-03-26:
Odd but good to know.  Enjoy

closed 2023-03-26 by christianhaitian