christianhaitian/arkos #678: Update check fails issue

Dantasstic, 2023-03-26:
### Describe the Issue (If applicable)
Checking for RA updates on RG351V produces an error screen that does not register any inputs to close the screen, and no updates occur. 

Error states: _"Looks like OTA updating is currently down or your wifi or internet connection is not functioning correctly."_ There is a highlighed on-screen button that says OK but you cannot use any button to make this screen go away. The device must be powered down by holding the F button and pressing power.

WiFi connection is working, as I am able to update cores and such from within RetroArch

### How can the issue be reproduced? (If applicable)
Be running ArkOS 2.0 (03112023) on a RG351V and try to run Update under Options.

### What device are you using?
RG351V

### What version of ArkOS are you on?
ArkOS 2.0 (03112023)

### Anything else you'd like to include that may help to review this issue or feature request?
[image]

closed 2023-03-26 by Dantasstic

christianhaitian, 2023-03-26:
https://github.com/christianhaitian/arkos/issues/661

Dantasstic, 2023-03-26:
> #661

Yeah, I saw that after posting, which is why I closed it. It is concerning though that the error wouldn't let you click OK and trapped you on that screen which out forcing a shutdown. I'm guessing you fixed that as well in the newest update, though.

christianhaitian, 2023-03-26:
That shouldn’t have happened but hopefully this solution addresses that for you too.