christianhaitian/arkos #679: Request: Support for Powkiddy RK2023

banana3333, 2023-03-26:
Probably a longshot, but is there any chance of support for RK2023 being added? Mine was able to boot the RG353V/VS image, but the display seems to be vertically offset downwards a bit. A big row at the top is empty while the bottom is cut off. The battery reader also understandably isn't working here and just reads 0% no matter the charge.

[image]
[image]

Alternatively, if there is a way I could adjust the vertical offset with the current image that would be a huge help.

christianhaitian, 2023-03-26:
I've requested a couple of development units from Powkiddy and they're supposed to be sending me some.  We'll see.

christianhaitian, 2023-03-27:
Just heard JelOS may add official support for this unit as well.  Will depend on if they can get some units which they're working on.

In the meantime, can someone post the contents of the boot folder in a zip file here.  Mostly looking for the .dtb file(s).  Thanks in advance

christianhaitian, 2023-03-27:
> The JELOS partition had a .dtb file, is this maybe what you wanted?
> 
> [image]
> 
> https://www.mediafire.com/file/4ec67kym04enpha/JELOS.zip/file EDIT: Also here is the full image if you want that https://mega.nz/file/HcgH3AZA#8SWbKfeW4rxPglmdMYUgqZFapHboFBoiTTHnwpDs3Mg
> 
> I found a boot folder under the root partition, but this Linux Reader program I'm using to view it on Windows is telling me nothing is in there.
> 
> [image]

That dtb is what I was looking for.  Thanks for providing that jelos.zip.  Attached is a zip file that contains an updated rk3566-OC.dtb file.  Try replacing the one in the boot partition on the ArkOS card with this one and see if that changes anything.  Let us know your results.

[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/11083244/rk3566-OC.zip)

banana3333, 2023-03-27:
Awesome! That did the trick! Thank you Christian!

[image]

christianhaitian, 2023-03-27:
Nice!.  Enjoy!

christianhaitian, 2023-03-27:
Hopefully all the buttons works correctly.

banana3333, 2023-03-27:
Yeah I figured it would be easier for you to work on stuff like battery when you actually have a device. All the buttons are working perfectly.

banana3333, 2023-03-28:
> > Yeah I figured it would be easier for you to work on stuff like battery when you actually have a device. All the buttons are working perfectly.
> 
> Is this with or without using the new dtb file.. Just curious

This is with using the new dtb file.

christianhaitian, 2023-03-28:
Hi, can someone load this updated rk3566-OC.dtb file on to this unit and see if battery reading works now and report the results.  Thank you.
[rk3566-OC.dtb.zip](https://github.com/christianhaitian/arkos/files/11094767/rk3566-OC.dtb.zip)

banana3333, 2023-03-29:
I gave it a shot on mine and the battery reading was showing 0% again. It still works just fine otherwise though!

christianhaitian, 2023-04-06:
Updated kernel and dtb that fixes the battery reading and analog stick for this device.  Thanks to @fewtarius for the guidance on the battery fix and TSS_Killer for confirming the analog and battery fixes for this device.

[RK2023-AnalogandBatteryFix.zip](https://github.com/christianhaitian/arkos/files/11174803/RK2023-AnalogandBatteryFix.zip)

christianhaitian, 2023-04-09:
An image is now available for the RK2023 from the wiki [here](https://github.com/christianhaitian/arkos/wiki#download-links).  Closing this request as this device now officially supported by ArkOS.

closed 2023-04-09 by christianhaitian