christianhaitian/arkos #680: Issues with V2 screen on RG353V and not booting properly!

BamBunBam, 2023-03-28:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

Booting on the RG353V with the V2 screen gives blank or white screen. Editing bios rk3566-OC.dtb to rk3566-OC.dtb.v1 and rk3566-OC.dtb.v2 to rk3566-OC.dtb to be able to see boot and then it freezes.

### How can the issue be reproduced? (If applicable)

First boost for ArkOS without editing rk3566-OC.dtb will show the big issue on V2.
Editing bios on first or second boot locks you at the ARKOS splash screen.

### What device are you using?

RG353V Screen V2

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOS 2.0 (03112023)

### Anything else you'd like to include that may help to review this issue or feature request?

Anbernic stock OS boots as well as android without consequence. I would prefer ArkOS as you have created a beautiful software and quality of life.

Thank you in advance for any consideration or assistance.

RetroGFX, 2023-03-28:
I think u may have had an issue flashing.. Reflash, then change the dtb BEFORE u boot your handheld with the sd in it

BamBunBam, 2023-03-28:
> I think u may have had an issue flashing.. Reflash, then change the dtb BEFORE u boot your handheld with the sd in it

I have actually done it both ways to see if it made a difference and it does not unfortunately. I see some folks get lucky and get it working and others don't and I think I am one of the unlucky ones.

BamBunBam, 2023-03-28:
To add to this, just did another round of flashing and editing bios to try again and this time, the ArkOS splash came up and then over it the version and it froze 2 seconds later on the installation it seems like. Been frozen for 2 minutes now.

christianhaitian, 2023-03-28:
Sounds like a possibly bad sd card.  What sd card make are you using?

BamBunBam, 2023-03-28:
Sandisk SDXC 128gb A2

[image]

christianhaitian, 2023-03-28:
have you made sure your unit is fully charged?

BamBunBam, 2023-03-28:
> have you made sure your unit is fully charged?

Wait, is that really a thing?

christianhaitian, 2023-03-28:
very much so.  If the device is getting low on power, lock ups can happen pretty frequenly due to not have enough power to complete certain tasks.

BamBunBam, 2023-03-28:
> very much so. If the device is getting low on power, lock ups can happen pretty frequenly due to not have enough power to complete certain tasks.

It's currently at 80%. I will charge it fully and try again and report back. Just to be safe I will also reflash the sd card.

RetroGFX, 2023-03-28:
U could try this also and see if it works
https://drive.google.com/file/d/1W0xL2E5u6cAeoZj2abA6wD3aflGeL6dD/view?usp=drivesdk

RetroGFX, 2023-03-28:
It's a v2 image of UnofficialOS for the 353v.. Just something to check to see if you have images with that also.. Christian and I work together a lot so this is basically just trouble shooting for ya

BamBunBam, 2023-03-28:
> It's a v2 image of UnofficialOS for the 353v.. Just something to check to see if you have images with that also.. Christian and I work together a lot so this is basically just trouble shooting for ya

I hope I didn't sound defensive or like an ass on that reply. I just wanted to know what I was installing before I did it. Thank you for your help guys!

RetroGFX, 2023-03-28:
No not at all bud your good

BamBunBam, 2023-03-28:
> U could try this also and see if it works https://drive.google.com/file/d/1W0xL2E5u6cAeoZj2abA6wD3aflGeL6dD/view?usp=drivesdk

Is this the same process as ArkOS?

RetroGFX, 2023-03-28:
This image just needs to be flashed if u def have a v2 screen.. Is it's a v1 it's a separate image

BamBunBam, 2023-03-28:
I am using rufus per retro game corps. This unofficial worked!!!

christianhaitian, 2023-03-28:
Try a different sd card.  Could be this one is faulty.  It's rare for sandisk cards but certainly possible.

BamBunBam, 2023-03-28:
I actually tried my 256gb as well and it didn't work for Arkos but this unofficial is doing okay

christianhaitian, 2023-03-28:
Ok.  Use that.  The OS are similar in performance.  Just menu options are different.

BamBunBam, 2023-03-28:
> Ok. Use that. The OSes are similar in performance. Just menu options are different.

Will do thank you

closed 2023-03-28 by christianhaitian

aims17, 2023-07-26:
Heyyy guys! I am also having this kind of issue on my 353V, 

Installed ArkOs then it was just stuck on tyhe ArkOS screen. 

Huhuhu what should I do? Please help. The link you've provided above no longer exists. 

I really want my 3b3V have ArkOS please do help and advise me

christianhaitian, 2023-07-26:
Use a Sandisk or Samsung sd card.

aims17, 2023-07-26:
Hi Christian,

yes i am currently using an original Sandisk 16GB MicroSD card and it still wont work :(

christianhaitian, 2023-07-26:
You have another sd card you can try?

aims17, 2023-07-26:
Yes, a bnew original sandisk microsd 256gb

aims17, 2023-07-26:
Though it still wont work :(

christianhaitian, 2023-07-26:
What program are you using to image the sd card?

aims17, 2023-07-26:
I tried both Rufus and Win32 disk imager

christianhaitian, 2023-07-26:
Try USB IT per the wiki and make sure your device is sufficiently charged.

aims17, 2023-07-26:
Will do try it now thankss!