christianhaitian/arkos #684: Access to Gamecube settings - RG353M - ArkOS

paladinperry, 2023-03-29:
Evening Christian.

I've had a scout on the wiki and reddits , tried all button inputs. is there something I am not doing correctly ?
I'm trying to access the settings from within a gamecube rom to try frameskip

RG353M latest ArkOS revision - It's managed surprisingly well in a few titles and I want to try messing with frameskip settings etc
Thanks again for all the hard work, I'm using ArkOS exclusively now

christianhaitian, 2023-03-29:
The gamecube emulator is dolphin standalone so there are no settings to access from a menu like retroarch.  frameskip is already on but if you'd like to change settings, you can do so via ssh by modifying `/home/ark/.local/share/dolphin-emu/Config/Dolphin.ini`.  You can also try changing some options in `/home/ark/.local/share/dolphin-emu/Config/GFX.ini`.  If you find some setting changes that seems beneficial, feel free to share here.  Thanks,

closed 2023-03-29 by christianhaitian

paladinperry, 2023-03-30:
thankyou Christian I will continue testing (mainly 2d) and I will post configs for the arkOS crew once finished !