christianhaitian/arkos #685: How to connect an audio device to  bluetooth headphones

fpiesik, 2023-03-30:
This is not an issue but a question. I'm sorry, but I don't know were else to ask. I even tried chatGPT to help me on this:)

So I use the rg353m as my [m8-Tracker](https://dirtywave.com/), a music creation device that's inspired by [lsdj](https://www.littlesounddj.com/lsd/index.php). There is a headless version that runs on a [Teensy 4.1](https://www.pjrc.com/store/teensy41.html), a powerful microcontroller board. This can be plugged in to the usb-c-otg port on the rg353m and used by an [interface software](https://github.com/jasonporritt/rg351_m8c) that is to be installed on the ports folder.

It works really well when running it with the built in speakers but sadly not with bluetooth audio. My qustion is how to connect an external audio device (in this case the teensy-board) to the bluetooth output? 
The script that starts the software uses this command to make the connection from the audio device to the speakers:

`alsaloop -P hw:0,0 -C hw:2,0 -t 200000 -A 5 --rate 44100 --sync=0 -T -1 -d`

when I use aplay -l to to list all audio outputs, it lists two loopback devices that are created when bluetooth is activated numbered card 2. But changing the playback hardware in that command above to 2 like this:

`alsaloop -P hw:2,0 -C hw:2,0 -t 200000 -A 5 --rate 44100 --sync=0 -T -1 -d`

just leads to not playing any sound at all. I've tried many different versions of this command with no luck. So I guees my approach is wrong somehow. 

I would be very thankful to get a hint on how to approach this problem!

christianhaitian, 2023-03-31:
Have you paired this device with ArkOS via the Bluetooth script in the options section?

fpiesik, 2023-03-31:
Yes I did. So in general the approach of connecting to those loopback devices via the alsaloop-command is fine, right? 
In order to prevent the loopbacks to be connected to something else, I tried to change their name by editing the ~/.asoundrcbt file. But no matter how I changed the names there, it had no impact to the actual generated loopback devices.

christianhaitian, 2023-03-31:
Hi unfortunately, I'm not sure how best to help on this one.

fpiesik, 2023-03-31:
Okay, thank you!

closed 2023-03-31 by fpiesik