christianhaitian/arkos #687: Missing options for RG353V

JackGrbovic, 2023-04-08:
On the RG353V, I'm trying to use the M8 headless through some scripts (https://github.com/jasonporritt/rg351_m8c) in the ports folder in ArkOS. This is basically a Teensy 4.1 device connected through the top USBC slot, and it almost works quite well. However it seems that certain aspects of ArkOS on this device in particular are stopping me from getting the functionality I need.

If I boot the device with the Teensy plugged in, the RG353V seems to default to the Teensy as the only audio output, and since, on the RG353V, there is no option in the sound settings to change audio output (though this seems to be present on other Anbernic models), I cannot get any audio, not in any games or while using the M8 headless itself. I have tried the F + L3 shortcut many times, with no success, but this does work when the devices is started without the Teensy, then the Teensy is plugged in, and the M8 headless can be used - until it crashes. The pulseaudio version provided is said to be a good alternative to the standard alsaloop one (with regard to crashes), but I get no audio in that whatsoever, so I tried to change this in ssh, where I encountered more issues.

I thought I'd see if I could change the audio output device through ssh, however the FAQs for ArkOS says to go to Options and then to Network Info. I do not have this option, it just isn't there. Luckily if I press start on the Emulation Station menu, I can see my device's IP there, however, when I try to connect in ssh, and it asks me for a password (the default in the FAQs is said to be 'ark'), and I give it 'ark', it says it's not correct (I have enabled remote services). I tried changing this in Emulation Station, but I need the password to do it.  I have a strong suspicion that this would be visible to me in the Network Info section, though I have no access to it, so cannot see the password.

Basically the main issues here are:
- Audio output defaults to USBC device (which isn't an output) and results in no sound at all
- This cannot be changed, not with a hotkey, by changing the volume, or in the menus (as far as I can tell)
- Trying to work around this with ssh but with no Network Info option in Options, results in no ability to know the password (which is said to be set to 'ark' by default, but when I provide this, it is incorrect) and gain access through ssh

christianhaitian, 2023-04-08:
Unless you've changed the password or downloaded the ArkOS image from some other location that's not listed on this wiki, the user and password is indeed ark and ark.  Nothing more I can help with on this since it does work fine on my end.

closed 2023-04-08 by christianhaitian

JackGrbovic, 2023-04-08:
I'll try a reinstall. So what I've said should mostly not be the case then right?

christianhaitian, 2023-04-08:
For logging into ArkOS via ssh, that's correct.  the default username and password is `ark` for both.