christianhaitian/arkos #688: RG353M Vibration on device, not on controller

mega-joe, 2023-04-12:
### Describe the Issue (If applicable)
I just picked up an 8BitDo SN30 pro for my RG353M. It works, pretty much out of the box, which is great. The one exception is the vibration/rumble. When using the 8bitDo, the RG353M continues to vibrate via the built-in controller vibration (as if the built in controller is Port 1 Control, even though the 8bitDo is properly set and working on Port 1). The 8BitDo vibration is never engaged.

### How can the issue be reproduced? (If applicable)
I've tried this with both Dreamcast and PSX. If I don't use the 8BitDo, the built-in vibration works fine. If I use the 8BitDo as the controller, then the built-in vibration still happens, but on the 353M and not on the controller.

Controller is attached via bluetooth in the X-Input mode

### What device are you using?
RH353M
8BitDo SN30pro

### What version of ArkOS are you on?
ArkOS v2.0 (04/01/2023) build

### Anything else you'd like to include that may help to review this issue or feature request?
Apologies if this is just a configuration thing I missed.

christianhaitian, 2023-04-12:
with how vibration is done on this device currently, this is understandable.  Not sure if vibration to external controllers will ever work.   I don't have a solution for this at this time.

mega-joe, 2023-04-12:
Bummer. Thanks for the quick response -- and for all the great work you do for this project.

closed 2023-04-12 by christianhaitian