christianhaitian/arkos #689: Issue: can't "press" OK on update screen on RG351v

gfvictor, 2023-04-13:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

When I open the 'Update' from Options I'm led to a new screen where it warns me of the possible troubles of updating and prompt me to confirm by pressing the OK button. The problem is that none of the buttons seems to work on that screen, same goes on the 'Wifi' and 'Network Info'.

### What device are you using?

RG351V (new batch, with the V2 Screen, if it matters).

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

2.0

### Anything else you'd like to include that may help to review this issue or feature request?

Already tried [https://github.com/christianhaitian/arkos/issues/652#issuecomment-1446652500](url), but buttons still not working...

christianhaitian, 2023-04-13:
If this is a unit with a v2 screen, does audio work on your unit?

christianhaitian, 2023-04-13:
The issue you're having is related to that file you had to change.  It's not fully compatible with ArkOS is the issue.

gfvictor, 2023-04-13:
I see. Well, that’s ok then. It’s just a minor issue anyway. Thank you for your time and for the OS! Wish you the best

closed 2023-04-13 by gfvictor