christianhaitian/arkos #691: Add FBNeo to NeoGeo CD emulator list

zomgugoff, 2023-04-14:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ? Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions? Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? Yes

### Describe the Issue (If applicable)
FBNeo can be used to run NeoGeo CD games and is a little more accurate than Neocd, but has slower load times. It requires the "--subsystem neocd" option be added to the command which is a little difficult to implement when neocd does not take the "--subsystem" option. I have created a command line that works.

### How can the issue be reproduced? (If applicable)
It's a feature request

### What device are you using?
RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (04012023)

### Anything else you'd like to include that may help to review this issue or feature request?
Here is the modified /etc/emulationstation/es_systems.cfg entry I came up with:

        <name>neogeocd</name>
        <fullname>SNK - Neo Geo CD</fullname>
        <path>/roms2/neogeocd/</path>
        <extension>.cue .CUE .chd .CHD .m3u .M3U</extension>
        <command>if [ %CORE% = fbneo ]; then option="--subsystem neocd"; else option=""; fi; sudo perfmax %EMULATOR% %CORE%; nice -n -19 /usr/local/bin/retroarch -L /home/ark/.config/retroarch/cores/%CORE%_libretro.so $option %ROM%; sudo perfnorm</command>
           <emulators>
              <emulator name="retroarch">
                <cores>
                  <core>neocd</core>
                  <core>fbneo</core>
                </cores>
              </emulator>
           </emulators>
        <platform>neogeocd</platform>
        <theme>neogeocd</theme>
        </system>

christianhaitian, 2023-04-15:
Cool.  I'll include it in the next update.  Thanks for the suggestion and research on this.

RetroGFX, 2023-04-15:
U gonna make this default, or just add it in as an emulator option

zomgugoff, 2023-04-15:
I would prefer it as default, but I don't know that others would like the slower load times.

RetroGFX, 2023-04-15:
> I would prefer it as default, but I don't know that others would like the slower load times.

Yea I hear ya.. I'm sure Christain will use his judgement and use whichever he feels is the better default option..I get it's more accurate, but if say, there's less game compatibility, then it's really not better from a developer perspective to use it as default, ya know.. This is all stuff I'm sure he'll look at before he decides..I work with him daily, all of our den decisions are usually always thoroughly thought thru.

christianhaitian, 2023-04-15:
I will not use this as a default.  it seems more picky about acceptable neo geo cd games.  I have yet to get any to load successfully with fbneo that load with no problem with neocd.

christianhaitian, 2023-04-15:
I had to use either a neogeo romset that has .cue/.ccd/.img/.sub or a single .bin and .cue file for fbneo to take it.  That's a little annoying to me but it works.  I'll need to update the wiki to note that extra steps for fbneo to work.

RetroGFX, 2023-04-15:
> I had to use either a neogeo romset that has .cue/.ccd/.img/.sub or a single .bin and .cue file for fbneo to take it. That's a little annoying to me but it works. I'll need to update the wiki to note that extra steps for fbneo to work.

Nice, good to know.. I should look into what we're using on uOS as far as neo CD.. Might be an idea for us to add this as an option also . Nice work buddy

zomgugoff, 2023-04-16:
I have been using the Redump set and that is in (multi-)bin/cue. I didn't know it doesn't accept .iso files.

christianhaitian, 2023-04-16:
I couldn't get any multi bin/cue files from redump to work.

zomgugoff, 2023-04-16:
Gah, you're right. They don't work. I forgot I did conversions to single bin/cue on the multi-bin games.

RetroGFX, 2023-04-16:
Ya man, can't have it default with all the quirks it has.. Default needs wide hand compatibility from a devs support point of view

christianhaitian, 2023-04-29:
This has been accommodated with the latest update.

closed 2023-04-29 by christianhaitian